damellis/ESP #5: Label chart axes (w/ numeric values).

damellis, 2016-01-22:
For the raw data, pre-processed data, features, and training data.

nebgnahz, 2016-01-22:
For Y-axis, this task is pending on [a pull request to ofxGrt](https://github.com/nickgillian/ofxGrt/pull/2).
For X-axis, not sure what to display.

nebgnahz, 2016-01-22:
`Good to know... I think we may want to create our own fork of the GRT for development. We should still try to send changes upstream but I don't want to get stuck waiting for them to be merged. What do you think?`

Replying here:
Agree with our [own fork](https://github.com/nebgnahz/ofxGrt).

damellis, 2016-01-23:
Great. I added you to my fork of the main GRT here: https://github.com/damellis/grt/

nebgnahz, 2016-04-16:
this issue is resolved...

closed 2016-04-16 by nebgnahz