damellis/ESP #6: Don't hardcode choice of stream type.

damellis, 2016-01-22:
Right now, we're hardcoding the type of stream by instantiating a specific class in ofApp::setup(). We're going to want some way for the user to pick between different stream types, either in their code or in the GUI. For instance, you might want to try out an audio pipeline with the computer's microphone or with an mic connected to an Arduino streaming to the computer.

nebgnahz, 2016-01-22:
Originally I was planning to have the code in `user.h`, however I faced a bit C++ structure challenge:
`ofApp` should store the base class, thus only a pointer; and the will user needs to pass in an instantiated object. The code would look like:

```
void setupInputStream(std::unique_ptr<IStream> is) {
  is.reset(new AudioStream());
  is->useAnalogPin(0);
  [ . . . ]
}
```

And it seems a bit clumsy. Let me know if you have a better way.

damellis, 2016-02-01:
Changed to have the user define the stream and pipeline as globals and register them with the ofApp.

closed 2016-02-01 by damellis