damellis/ESP #32: Don't require external compilation of the GRT.

damellis, 2016-02-09:
We should either bundle a compiled version (as in newer version of the ofxGrt addon) or include the GRT in the Xcode project so that it's built automatically.

nebgnahz, 2016-02-10:
We could optionally provide a shell script like `setup.sh` to ease the process of setting things up.

Currently users need to do submodule update and create a soft link for `ofxGrt`.

damellis, 2016-04-23:
I think I fixed this. Note that to get the application we build to correctly link against the libgrt.dylib in its Frameworks/ directory, I needed to set the id of libgrt.dylib:

`install_name_tool -id @executable_path/../Frameworks/libgrt.dylib Xcode/SmartSensors/libgrt.dylib`

Apparently, when an executable is linked against a dynamic library, it remembers the id of the library and searches for it in that path when run.  See: https://github.com/conda/conda-build/issues/279

damellis, 2016-04-25:
Closing for now. Note that we still need the GRT header files, so we should probably keep the GRT submodule. If we need to update the GRT regularly, we may want to switch to compiling the GRT as an Xcode project / target which can be a dependency of our main project.

closed 2016-04-25 by damellis