damellis/ESP #34: Rename Audio Xcode project.

damellis, 2016-02-10:
To something more descriptive. Maybe just "sensors"? Or SmartSensors?

nebgnahz, 2016-02-10:
Maybe the following:

Current
```
Xcode
├── Audio
│   ├── Audio.xcodeproj
│   ├── Makefile
│   ├── Project.xcconfig
│   ├── addons.make
│   ├── bin
│   ├── config.make
│   ├── obj
│   ├── openFrameworks-Info.plist
│   └── src
└── GUI
    ├── GUI.xcodeproj
    ├── Makefile
    ├── Project.xcconfig
    ├── addons.make
    ├── bin
    ├── config.make
    ├── openFrameworks-Info.plist
    └── src
```

Proposed
```
Xcode
├── SmartSensors
│   ├── SmartSensors.xcodeproj
│   ├── Makefile
│   ├── Project.xcconfig
│   ├── addons.make
│   ├── bin
│   ├── config.make
│   ├── obj
│   ├── openFrameworks-Info.plist
│   └── src
```

damellis, 2016-02-10:
That is, get rid of the current "GUI" project and rename "Audio" to "SmartSensors"? Sounds good to me.

nebgnahz, 2016-02-11:
Done.

- Remove "GUI" https://github.com/damellis/sensors/commit/fd34c8cb7e0a5edf5ffee2b2eb81cd14dfe4ec29
- Rename "Audio" to "SmartSensors"
https://github.com/damellis/sensors/commit/17319c65bcd3c3683308e210068da468fd5a2191

closed 2016-02-11 by nebgnahz