damellis/ESP #38: Ability to name a training sample.

damellis, 2016-02-11:
So that we don't have to remember what's what when reloading old training sets.

nebgnahz, 2016-02-17:
`GRT::TimeSeriesClassificationData::setClassNameForCorrespondingClassLabel` should work; but challenging on UI side: `ofApp` will trigger training on `t`, loading on `l`, etc.

nebgnahz, 2016-02-18:
Done.

- https://github.com/damellis/sensors/commit/33b516601a1ca3086ac4495aba4fa16bdcadd74c
    Use previous name after loading existing data.

- https://github.com/damellis/sensors/commit/ad16d06189a2bfb033e091dfbc529bf4276a7d1b
    use display title to blink

- https://github.com/damellis/sensors/commit/7384469a5c3cb05bb96773c1abb1488f6dfc07dc
    initial implementation of rename samples

closed 2016-02-18 by nebgnahz