damellis/ESP #46: Training class labels not saved with training data?

damellis, 2016-02-18:
When I record training samples, give them names, and then save them, the names don't seem to be stored in the .grt file. (Attached is an example, where the classes should have been named "Foo" and "Bar".)

[TrainingData-test.grt.txt](https://github.com/damellis/sensors/files/137068/TrainingData-test.grt.txt)

nebgnahz, 2016-02-18:
The GRT library doesn't seem to store it.

https://github.com/nickgillian/grt/blob/0f4b39e96b121f103592942f09b4a11fe572fbf6/GRT/DataStructures/TimeSeriesClassificationData.cpp#L362

We should make changes to your [fork](https://github.com/damellis/grt).

nebgnahz, 2016-02-19:
Fixed with https://github.com/damellis/grt/commit/4177f2a082b386878b765d449c40806efdbd4877.

closed 2016-02-19 by nebgnahz

nebgnahz, 2016-02-19:
One more comment: you will need to update GRT (by recompiling it...).

So we should think about https://github.com/damellis/sensors/issues/32.