damellis/ESP #47: Problems saving and loading training class names with special characters.

damellis, 2016-02-20:
I'm having trouble saving and loading training data with training class names. For instance, when there's a capital letter in the name, it seems we store characters for the shift key presses (00 02) in addition to (and preceding) the capital letter itself. See attached.
 
[TrainingData-test.grt.txt](https://github.com/damellis/sensors/files/138896/TrainingData-test.grt.txt)

damellis, 2016-02-20:
Also, are spaces in the names saved and loaded correctly?

nebgnahz, 2016-02-20:
Only add normal character instead of every key stroke.
(https://github.com/damellis/sensors/commit/305190974e098d60f402e1c94b055e6a2a01abab).

Shift key issue should be fixed.

Space is saved correctly, but loading doesn't work.

damellis, 2016-02-20:
Hmm... we could just ignore space keys too (not allow them in the class names).

nebgnahz, 2016-03-01:
Addressed in https://github.com/damellis/sensors/commit/f01e157faef68cfb84c9ba7a21e289975fdeb4fe.

Spaces are allowed but will be remapped to `_`.

closed 2016-03-01 by nebgnahz