damellis/ESP #48: Way to show features in training tab.

damellis, 2016-03-01:
For some data (e.g. audio), it's hard to tell if you've got a good sample by looking at the raw input data. Would help to have a way to show features, too.

nebgnahz, 2016-03-03:
This gets complicated for two cases:
1. Audio FFT features are in high dimension (512 for example). For a sample, we cannot show the whole in a time-series plot; and we also shouldn't just show the FFT of a single frame. Ideally, a gif/video like visualization would be most compelling; but the implementation could take a longer time.
2. For other features, since features are not comparable to each other, naively putting them in a single plot may not make sense. For example, mean value might always be 128 for a sensor reporting from 0 to 255 while the stdev is around 0 when idle; the scale (y-axis) wouldn't make much sense when directly using a time-series plot.

I need to think more about "show features".

damellis, 2016-03-04:
One thought for the non-FFT case: what about stacking the features vertically (i.e. within the same space used for the raw input sample)? That would place each features from sample 1, say, directly next to the corresponding feature from sample 2, for comparison between them.

nebgnahz, 2016-03-07:
Committed into master branch.

nebgnahz, 2016-03-16:
Closed as fixed (mostly).

nebgnahz, 2016-03-16:
Show FFT features as as in spectrogram.

damellis, 2016-04-18:
I'm going to call this done for now, since we do show the features on the training tab. Showing spectrograms could be nice, but that's probably a separate issue.

closed 2016-04-18 by damellis