damellis/ESP #58: Exception when showing features for pipelines without features.

damellis, 2016-03-07:
````
Thread 1Queue : com.apple.main-thread (serial)
#0	0x00000001000139ec in GRT::Matrix<double>::getNumRows() const at /usr/local/include/GRT/Util/Matrix.h:531
#1	0x000000010000a14f in ofApp::populateSampleFeatures(unsigned int) at /Users/mellis/Documents/Berkeley/Sensors/sensors/Xcode/SmartSensors/src/ofApp.cpp:295
#2	0x0000000100010717 in ofApp::toggleFeatureView() at /Users/mellis/Documents/Berkeley/Sensors/sensors/Xcode/SmartSensors/src/ofApp.cpp:836
#3	0x00000001000105aa in ofApp::keyPressed(int) at /Users/mellis/Documents/Berkeley/Sensors/sensors/Xcode/SmartSensors/src/ofApp.cpp:812
#4	0x00000001000155c1 in ofBaseApp::keyPressed(ofKeyEventArgs&) at /Users/mellis/Documents/Berkeley/Sensors/sensors/Xcode/SmartSensors/../../third-party/openFrameworks/libs/openFrameworks/app/ofBaseApp.h:80
#5	0x00000001005e3a6b in std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)::operator()(void const*, ofKeyEventArgs&) const [inlined] at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworksCompiled/project/osx/../../../../libs/openFrameworks/events/ofEvent.h:237
#6	0x00000001005e3a4d in decltype(std::__1::forward<ofBaseApp>(fp)(std::__1::forward<void const*, ofKeyEventArgs&>(fp0))) std::__1::__invoke<std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)&, void const*, ofKeyEventArgs&>(ofBaseApp&&, void const*&&, ofKeyEventArgs&&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__functional_base:415
#7	0x00000001005e3a4d in bool std::__1::__invoke_void_return_wrapper<bool>::__call<std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)&, void const*, ofKeyEventArgs&>(std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)&&&, void const*&&, ofKeyEventArgs&&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__functional_base:430
#8	0x00000001005e3a4d in std::__1::__function::__func<std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&), std::__1::allocator<std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)>, bool (void const*, ofKeyEventArgs&)>::operator()(void const*&&, ofKeyEventArgs&) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:1407
#9	0x00000001005c14d5 in std::__1::function<bool (void const*, ofKeyEventArgs&)>::operator()(void const*, ofKeyEventArgs&) const at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:1793
#10	0x00000001005c1306 in of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex>::notify(void const*, ofKeyEventArgs&) [inlined] at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworksCompiled/project/osx/../../../../libs/openFrameworks/events/ofEvent.h:136
#11	0x00000001005c12e7 in ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::notify(void const*, ofKeyEventArgs&) at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworksCompiled/project/osx/../../../../libs/openFrameworks/events/ofEvent.h:303
#12	0x00000001005c0154 in void ofNotifyEvent<ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>, ofKeyEventArgs>(ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>&, ofKeyEventArgs&) [inlined] at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworksCompiled/project/osx/../../../../libs/openFrameworks/events/ofEventUtils.h:222
#13	0x00000001005c0143 in ofCoreEvents::notifyKeyPressed(int, int, int, int) at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworks/events/ofEvents.cpp:251
#14	0x00000001003ddde1 in -[GLFWContentView keyDown:] ()
#19	0x00000001003df334 in _glfwPlatformPollEvents ()
#20	0x00000001005e2a51 in ofMainLoop::pollEvents() [inlined] at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworks/app/ofMainLoop.cpp:148
#21	0x00000001005e2a3f in ofMainLoop::loop() at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworks/app/ofMainLoop.cpp:122
#22	0x00000001005bc36c in ofRunMainLoop() [inlined] at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworks/app/ofAppRunner.cpp:173
#23	0x00000001005bc35f in ofRunApp(std::__1::shared_ptr<ofBaseApp>) at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworks/app/ofAppRunner.cpp:159
#24	0x00000001005bc2e4 in ofRunApp(ofBaseApp*) at /Users/mellis/Documents/Berkeley/Sensors/sensors/third-party/openFrameworks/libs/openFrameworks/app/ofAppRunner.cpp:153
#25	0x000000010000192c in main at /Users/mellis/Documents/Berkeley/Sensors/sensors/Xcode/SmartSensors/src/main.cpp:11
#26	0x00000001000018d4 in start ()
````

nebgnahz, 2016-03-07:
This should fix it:
https://github.com/damellis/sensors/commit/5567d3bbc4e91e563d2f09bffdea6dfa27b37e9a

nebgnahz, 2016-03-16:
Closed as no new updates on this.

closed 2016-03-16 by nebgnahz