damellis/ESP #65: Class names lost when adding new data points / re-training model (?)

damellis, 2016-03-07:
When I modify the training data and retrain the model, the class name no longer show up in the live pipeline view. Instead, "Label 1", etc are shown shown. The correct names are still shown in the training sample plots.
[image]

nebgnahz, 2016-03-08:
Deleting calls `training_data_.eraseAllSamplesWithClassLabel(label);`, therefore the class name is also lost.

The commit (https://github.com/damellis/sensors/commit/bf3e6a7af1a09a930e28977dea533408fef1636e) fixes this.

damellis, 2016-03-09:
I think this also happens when trimming samples.

closed 2016-03-09 by damellis