damellis/ESP #78: Show indication when training data has changed since the model was last trained.

damellis, 2016-03-10:
i.e. indicate a "edited" state requiring re-training, similar to showing files that have been edited and not saved.

nebgnahz, 2016-03-14:
Probably show this indication in the prediction (by graying the text?).

From the workflow, after adding new samples, people might expect the prediction results improved. At that stage, their attention is mainly on the prediction.

damellis, 2016-03-15:
I was thinking of something like inserting an asterisk (*) before the name of the class label, which I feel like is something I've seen in other editors. Or maybe the word "edited" after the label (which might mean on the next line)?

nebgnahz, 2016-03-16:
See https://github.com/damellis/sensors/commit/1db6f025edab24a9fa29a4ca7704b4465f6270ad.

Note that adding samples followed by deleting it is considered "edited".

closed 2016-03-16 by nebgnahz