damellis/ESP #86: Way to tune the post-processing parameters in the GUI.

damellis, 2016-03-10:
e.g. parameters for the ClassLabelFilter.

nebgnahz, 2016-03-21:
@damellis 
Generalized to pipeline parameter. 

For an example, check out https://github.com/damellis/sensors/commit/e3c1e2b86332dd5fdc1e4a63fc6b8a3898321084.

nebgnahz, 2016-03-21:
UI:
1. Type 'h' to open the "parameter tuning" panel.
2. Control the slider or click the button to change the value. Under the hood, a callback function `Tuneable::onXXXEvent` is invoked which will call `ofApp::reloadPipelineModules()`.

P.S. Shouldn't close this.... reopening.

nebgnahz, 2016-03-31:
Closing this issue as parameter tuning is implemented. And update to previous comment:
1. No need to type 'h'. The tuning panel is visible all the time, although collapsed by default.

closed 2016-03-31 by nebgnahz