damellis/ESP #87: Way to extract training samples (and test data) from history of live data stream.

damellis, 2016-03-10:
A way to pause or rewind the live data stream and then mark regions as new training samples (or false positives?). Perhaps also a way to store some portion of the history of the live data stream as test data to reuse later.

damellis, 2016-04-20:
Start with what's currently stored / plotted in the buffer, using the current pause key ('p').

nebgnahz, 2016-04-28:
Use `p` to pause; and use mouse to select, and press 1-9 to add the data to training data. Status texts are provided when you pause and make selection. You can only do this when the live stream is paused. 

We can add press 't' to add as test data. But for now, the live history data's size is pretty limited (because ofxGrtTimeSeriesPlot uses a circular buffer, we set it to be 256).

See implementation at https://github.com/damellis/sensors/commit/02e1d821e7738a4118be137073ab89e3d95d73fb.

closed 2016-04-28 by nebgnahz

damellis, 2016-04-28:
Awesome! Yeah, I think using this for test data will probably require a bigger re-imagining of the way we store (and navigate) the history of live data. So training samples only for now seems like the right way to go.