damellis/ESP #105: Explicit callbacks for tunable parameters (as alternative to automatic re-running of setup).

damellis, 2016-03-23:
Allow the example author to provide explicit code, in the form of a callback, to run when the value of a tunable is changed. Low priority.

damellis, 2016-08-02:
This is important for being able to adjust null rejection coefficients / thresholds for speaker identification without retraining the model.

nebgnahz, 2016-08-09:
Done with e37dbf2002cfec5538797f564245b8d0dcee42a9.

There are two possible behaviors when UI event happens:
1. If a corresponding callback is provided, it's called
2. If there is no callback provided, we proceed to reload the pipeline

For type, the callback should be a function with the signature 
```
template<typename T>
void f(T t);
```

The actual implementation can be improved with templates, but for now, we store the types and associated callbacks explicitly. See an example below.

```
int duration = 20;  // ms

void setup() {
    ...
    registerTuneable(duration, 0, 100,
                     "Timeout",
                     "How long (in milliseconds) to wait after recognizing a "
                     "gesture before recognizing another one.",
                     [](int new_timeout) {
                         std::cout << new_timeout << std::endl;
                     });
   ...
}
```

damellis, 2016-08-10:
One note is that we should generally update the global variable corresponding to the tunable as well as updating the relevant pipeline module. For instance, in your example, we'd want to set `duration = new_timeout`. That way, if another tunable causes the whole setup function to be re-run, the callback-enabled tunables will also be updated correctly.

nebgnahz, 2016-08-10:
The global variable is updated as the slider (UI) changes.

https://github.com/damellis/ESP/blob/master/Xcode/ESP/src/tuneable.cpp#L17

damellis, 2016-08-10:
Good point! :)

damellis, 2016-08-10:
I think I found a bug in the callback code. `Tuneable::onSliderEvent()` uses the callbacks associated with the `this` pointer, but since there's only one slider callback per GUI, we can't rely on `this` to point to the correct Tuneable. So we need to get the int_cb_ and double_cb_ from a Tunable instance derived from the event's target.

nebgnahz, 2016-08-10:
Yes, you are right. Thanks for the fix.

The whole `allTuneables` thing was a poor design as I was battling with the global variable and how to re-run `setup` function. We should refactor this in a cleaner way.

damellis, 2016-08-10:
Also made a small fix here: https://github.com/damellis/ESP/commit/d391d59c0c42f06ceec0ab76ea1395a2bae5fdee

damellis, 2016-08-10:
Let's call this fixed.

Some of the examples don't yet supply callbacks, but they should still work with the old system (calling setup()).

closed 2016-08-10 by damellis