damellis/ESP #113: Way to save values of tuned parameters.

damellis, 2016-04-02:
Not sure where these should go, though. In another file with a similar name to the training data maybe? e.g. if you save TrainingData.grt, you should also get TrainingData-parameters.grt?

damellis, 2016-04-20:
Use ofxDatGui saving / loading to start, if they have something reasonable.

nebgnahz, 2016-04-26:
Unfortunately their saving/loading is still under the roadmap (see http://braitsch.github.io/ofxDatGui/index.html#save-load-settings) and their description is copied below:

```
Save & Load Settings
The ability to save and load settings from an external file is currently in development.
Check out the project roadmap for more information.
```

nebgnahz, 2016-04-26:
The ability to support arbitrary ofxDatGui panel is going to be very involved. As of now, I have implemented a crude way for our purpose.

See https://github.com/damellis/sensors/commit/f7e9f07cf31dad7f25a11adc6b8ba1505f5b7b29.

Note the saved file has an ad-hoc data format `TYPE value`. Titles and descriptions are avoided as of now since user supplied title and description can contain almost arbitrary content (including spaces and new lines). Implementing our own parsing to cover these corner cases shouldn't be a priority.

closed 2016-04-26 by nebgnahz