damellis/ESP #131: first-time setup script

elsehow, 2016-04-12:
i think there should be a bash script you can run for first time setup

so, someone can just clone the repo and `./setup`

here's the script i made from the README directions (running now, will report back)

```shell
#!/bin/bash

git submodule init
git submodule update
ln -s $(pwd)/third-party/ofxGrt/ third-party/openFrameworks/addons/ofxGrt
ln -s $(pwd)/third-party/ofxDatGui/ third-party/openFrameworks/addons/ofxDatGui
ln -s $(pwd)/third-party/ofxParagraph/ third-party/openFrameworks/addons/ofxParagraph

cd third-party/grt/build
mkdir tmp
cd tmp
cmake ..
make
sudo make install
```

one thing i'd like to include is this tip from the readme:

`git submodule update --depth 1 -- third-party/openFrameworks`

this seems like a sensible default. 
how should it be incorporated into the script above?

and, does this sound like a good idea? shall i open a PR?

damellis, 2016-04-14:
That sounds good. Maybe including the openFrameworks depth 1 line before the regular git submodule update line will do the right thing? A pull request would be great!

damellis, 2016-04-15:
Merged!

closed 2016-04-15 by damellis

elsehow, 2016-04-15:
@damellis hooray! thanks. could you add some note to the readme about how to get up-and-running after setup?

damellis, 2016-04-15:
Will do.