damellis/ESP #136: Option to use features for all plots on training and analysis tabs.

damellis, 2016-04-18:
Currently, the "feature toggle" ("f") shows features in addition to input data on the training tab, but hides the live data stream. This is useful in allowing people to inspect the features for specific training samples but I think it would be good to support more general usage of features. That is, I can imagine situations in which someone wants to see features, not input data, everywhere: in the live data stream, in the visualizations of the training data samples, on the test data plots, etc. I think it would eventually be good to support this. 

(Note: this would probably require us to find a different / additional way of navigating through time for training samples with more than 32 features. I'm imagining some sort of timeline like that shown on the analysis tab, although I'm not sure what data it would show. A spectrogram would be nice, if possible. Otherwise, it could just be a slider with no data shown.)

damellis, 2016-04-18:
We may want to think about showing all features on the same plot, in different colors, as with the current input data.

damellis, 2016-06-01:
This might be useful for #215, for instance.

damellis, 2016-09-03:
This might also involve fixing #337.

For now, let's keep this simple:

1. for the live data view (on the analysis, training, and prediction tabs), let's show the bottom plot on the pipeline tab
2. for the training data view, let's keep all the interaction logic the same, and just change the UI layout. Specifically, when you toggle to feature mode, the live data plot should remain visible (although it will switch to feature mode as described in (1)) and the default training data plot (the one showing input data) should get shorter to make room for an adjoining feature plot. Let's keep the UI for navigating through and editing training samples in the same place (as the bottom of the interface).

nebgnahz, 2016-09-06:
Right now there is a divide by zero problem if there is not feature extraction modules.

nebgnahz, 2016-09-06:
There are cases that the pipeline doesn't have any feature extraction module but it does have pre-processing modules, in that case, show the last stage of pre-processing.

closed 2016-09-16 by damellis