damellis/ESP #138: Way to report warnings and errors for calibration samples.

damellis, 2016-04-18:
The example author should have a way to warn the user if a calibration sample seems problematic (e.g. lots of noise) or simply won't work at all (e.g. not enough data to perform the required computations). These should appear in the GUI, with the calibration sample not saved if an error is reported.

nebgnahz, 2016-04-18:
Related: https://github.com/damellis/sensors/issues/88

We need a way (small side panel for error reporting).

nebgnahz, 2016-05-26:
GUI part is easy, APIs for example authors are hard.

The current plan will be modifying the `CalibratorCallback` function signature such that the return value could either be success or failure with an error message.

nebgnahz, 2016-05-31:
Half-done with https://github.com/damellis/ESP/pull/217
Needs GUI support next.

nebgnahz, 2016-06-01:
Now GUI shows custom error message during calibration.

See commit https://github.com/damellis/ESP/commit/4144375e9be26d7fa23b69276bb638b4ca6faae2.

closed 2016-06-01 by nebgnahz