damellis/ESP #150: Simultaneous saving of all data?

damellis, 2016-04-18:
We may want a way to allow the user to save training data, test data, calibration data, and parameter values all at once. See #69, #113, #114.

damellis, 2016-04-25:
Removing this from the Cleanup milestone as I think we can get away without it for now.

damellis, 2016-06-01:
One way to do this would be to have the user specify the name of the directory to save or load, and have fixed file names within the directory for each type of data (training, test, etc.).

nebgnahz, 2016-07-17:
Commit https://github.com/damellis/ESP/commit/dee80cce5a0d2a9cb06951d79ada75a538feefdb makes the attempt to enable saving and loading all the data at the same time.

The abstraction is called an ESP session which includes calibration data, training data, test data and the pipeline (could have been trained).

The commit also unified the implementation: for every single type of data, we have `saveXWithPrompt`/`saveX` and `loadXWithPrompt`/`loadX`. `saveAll` and `loadAll` uses a pre-defined filename appended to a user selected directory name. All functions call `setStatus` which currently changes the status message so that the user will have an idea of what's going on.

`loadAll` and `saveAll` are bound to `L` and `S` keys.

nebgnahz, 2016-07-17:
Tunable parameters are not yet implemented.

damellis, 2016-07-18:
This all looks good. What do you think about making save all the default on exit?

damellis, 2016-07-26:
I think we can call this finished.

closed 2016-07-26 by damellis