damellis/ESP #158: Ability to specify the serial port in the GUI.

damellis, 2016-04-25:
I've been thinking that we might want to share specific examples / applications with people who don't want to install Xcode / compile everything from source. For example, it would be cool to write up a tutorial on doing gesture recognition using our DTW example. I think distributing a pre-compiled binary would basically work, expect that people probably need a way to select the serial (USB) port. 

Maybe this could be an integral tuning parameter? But we'd need to instantiate the ASCIISerialStream inside of the setup() function. Or maybe bring back a function for setting the serial port number.

nebgnahz, 2016-05-02:
I have a proposed solution in https://github.com/damellis/ESP/commit/76e9605fa2f131ca7f4acb937d587561b88872aa.

I think this shouldn't be a tuneable parameter because the example author wouldn't know the options (/dev/tty*) at compile time. This seems more to be a system support that our GUI should given special treatment.

I have a dynamic cast to detect the stream type (https://github.com/damellis/ESP/commit/76e9605fa2f131ca7f4acb937d587561b88872aa#diff-5e608dabd649c4d163addfe29cf61492R306).

damellis, 2016-05-02:
Wow, looks great! The basic functionality seems to work.

One problem is that selecting a serial port from the drop down menu for a second time triggers a failing alert in selectSerialDevice(). I'm not sure we need to support changing the serial port once one has been selected, but if we're not going to support it, we should probably remove the dropdown menu.

Also, I don't think it's obvious where the user would go to find the dropdown menu, especially if it's their first time using the software. Maybe the serial port selection should be a separate ofxDatGui menu? Or we should rename the current menu and be more explicit in the status message? 

A couple of more minor points: the status message about selecting a serial port should probably go away once you do select one. And for examples that require calibration, should we stay on the calibration tab until the user selects a serial port and collects the calibration samples?

nebgnahz, 2016-05-02:
Was fixing the few bugs but my Xcode project seems not happy---it keeps saying "failed to find ofConstants.h" although it is just there.

nebgnahz, 2016-05-02:
> One problem is that selecting a serial port from the drop down menu for a second time triggers a failing alert in selectSerialDevice(). I'm not sure we need to support changing the serial port once [...]

I have a patch that does precisely this. But needs setting up my openframework environment and test it.

> Also, I don't think it's obvious where the user would go to find the dropdown menu, especially if it's their first time using the software. Maybe the serial port selection should be a separate ofxDatGui menu? Or we should rename the current menu and be more explicit in the status message?

Agree. A separate GUI is fine but it increases the amount of management (for us and for the user). Maybe use the status text?

> A couple of more minor points: the status message about selecting a serial port should probably go away once you do select one. And for examples that require calibration, should we stay on the calibration tab until the user selects a serial port and collects the calibration samples?
Yes: it should go away and we should be on calibration tab.

nebgnahz, 2016-05-02:
Here are the patches...

[ofApp.cpp.patch.txt](https://github.com/damellis/ESP/files/245646/ofApp.cpp.patch.txt)
[ofApp.h.patch.txt](https://github.com/damellis/ESP/files/245647/ofApp.h.patch.txt)

You can either wait for an hour (for me downloading openFrameworks) or apply the patch (I think `git apply [path file]` should work).

damellis, 2016-05-02:
Thanks! I’m fine with waiting.

nebgnahz, 2016-05-03:
Ready to use after https://github.com/damellis/ESP/commit/f3e0b715c75fa7f759f671885334d807babeb5f3.

closed 2016-05-03 by nebgnahz