damellis/ESP #161: Tunable parameter text appears double size on larger monitors.

damellis, 2016-04-25:
My monitor is, it looks like, 2560 x 1600 pixels.

[image]

nebgnahz, 2016-04-28:
This is probably due to https://github.com/braitsch/ofxDatGui/blob/6dd5138801bb23a291dc0d11453e08dca6736b52/src/themes/ofxDatGuiTheme.h#L27.

Also I changed the default size https://github.com/nebgnahz/ofxDatGui/commit/4940fa52607d40aff9d9942d548b1b2ae119205e#diff-bbdd6e06a8e880afea8b672cf2a54ce7R216.

damellis, 2016-04-28:
> Also I changed the default size nebgnahz/ofxDatGui@4940fa5#diff-bbdd6e06a8e880afea8b672cf2a54ce7R216 <https://github.com/nebgnahz/ofxDatGui/commit/4940fa52607d40aff9d9942d548b1b2ae119205e#diff-bbdd6e06a8e880afea8b672cf2a54ce7R216>.

Hmm… I think openFrameworks does have a way to query the multiplier used for the screen (e.g. “1” for normal screens, “2” for retina). But this is not necessarily a high priority.

commit ce36cb59 (2016-09-30): Don't double ofxDatGui sizes on large displays.

Fixes #161.
closed 2016-09-30 by damellis