damellis/ESP #165: Support pipelines with no classifier for signal processing applications.

damellis, 2016-04-27:
There are times when you may not need a classifier but are just doing signal processing on an input signal. In such cases, I think our system still potentially offers a lot of value (visualizing of live sensor data, ability to connect to different input and output streams, etc.). It would be nice to support pipelines that don't have classifiers. We would probably also want to disable or hide the training tab, since there wouldn't be a need for training data. The analysis tab would probably need to change to showing the pipeline output rather than the prediction results, along with the test data (pipeline input).

damellis, 2016-04-28:
I added this to the "building more examples" milestone because I'm guessing that many of the examples we'll want to build will fall into this category.

nebgnahz, 2016-05-19:
Commit 20c6cd4356753cab2e4594164d9d07f4458b5128 attempts to do this.

1. Re-ordered the tabs so that hiding `Training` tab would be easier. A more modular approach would be implement a Fragment class (like Android) and then refactor the code; but that is a daunting task at this stage.

2. [user_sudden_motion.h](https://github.com/damellis/ESP/blob/20c6cd4356753cab2e4594164d9d07f4458b5128/Xcode/ESP/src/user_sudden_motion.h) detects the zero-crossing of derivatives (i.e. changes). With this example, if you are moving the accelerometer slowly, nothing will be reported; but if you are moving it fast, i.e., sudden motion, the zero-crossing will detect it.

There seems to be a general problem: the output dimension will not necessarily be one in contrast to classification results, which is an integer from 1 to `NumClass`.

damellis, 2016-05-19:
My thought is to create a separate output stream interface for multi-dimensional floating point output, e.g.

```
class OStreamVector : public virtual Stream {
public:
  virtual void onReceive(vector<float> data) = 0;
}
```

This would allow some output streams (e.g. TCP or ASCIISerial) to support both classifier output (uint32_t) and signal processing / regression output (vector<float>) while others (like keyboard and mouse) could continue to support only classifier output.

What do you think?

damellis, 2016-05-19:
Another thought is that we may sometimes want a pipeline whose output is a label / integer but that doesn't require training data or training. Kind of a pre-defined classifier (e.g. to distinguish talking vs background noise, or beat vs. no beat, etc). This would use the same kind of output streams as a classifier but would involve hiding the training tab, etc.

nebgnahz, 2016-05-19:
The term "classifier" is ambiguous. We have two different classifiers:
1. Classifiers that need to be trained (what we have today); this is a transformation function with unknown parameters.
2. Classifiers whose job is to classify data into a category (uint32_t) but the transformation function is pre-defined. No need to train.

The second one, generalized in a mathematical point of view, is no different from pre-processing, feature extraction or post-processing, except the output domain is one-dimension integer.

This is stepping into GRT's code structure and even semantics of processing pipeline. Help or discussion with Nick would be helpful.

damellis, 2016-05-20:
Sorry, maybe I was overthinking this... for the case where there's no real classifier (i.e. no training), let's just stick with float vector output.

closed 2016-05-23 by nebgnahz

nebgnahz, 2016-05-25:
I added a [FeatureApply](https://github.com/damellis/grt/commit/317b9ce568b5e54f6e992a93beb06201319bcd96) module that will apply a function during the feature extraction process.

One example can be found here: https://github.com/damellis/grt/blob/317b9ce568b5e54f6e992a93beb06201319bcd96/examples/FeatureExtractionModules/FeatureApplyExample/FeatureApplyExample.cpp

This should be generic enough to build other simple feature extraction module (needs to be stateless, only applying the function). Managing states is another complicated task.