damellis/ESP #169: Document our API (e.g. streams, calibrators, tunable parameters, and useXxxx() functions).

damellis, 2016-04-28:
Maybe as doxygen comments within the source code?

nebgnahz, 2016-04-28:
doxygen configuration is created (https://github.com/damellis/sensors/commit/8bf7ba7044a425dcbc9bd9289b09ed206e74ae7d).

Inside `Xcode/ESP/`, simply run `doxygen` to get the documentation inside `Xcode/ESP/` folder. We need a place to host the documentation.

A common way people do it with Github is to create a `gh-pages` branch and pushes the documentation there. I did it (https://github.com/damellis/sensors/commit/2a95facf880983e0a6b24223fa7254780924cf09) as well. So our initial documentation is available at https://damellis.github.io/sensors/.

nebgnahz, 2016-04-28:
This issue would then require modifying the source code to follow `doxygen` requirement and update the documentation.

damellis, 2016-04-29:
Awesome. GitHub pages seems like a good place.

closed 2016-05-26 by damellis