damellis/ESP #178: git clone of openFrameworks needs to grab commit referenced by submodule.

damellis, 2016-05-02:
Otherwise, you can't actually checkout the right commit because it's not present.

nebgnahz, 2016-05-03:
Short answer: it's not supported natively (or there is not an easy solution as of now).

Long answer: Currently submodule tracks a commit (a SHA1 specified snapshot of the repository). It is technically a detached head. Unless you cloned the full history and then checkout that specific version, shallow cloning won't have a way to trace from master back to that commit. According to this [Stackoverflow](http://stackoverflow.com/questions/2144406/git-shallow-submodules/17692710#17692710) discussion:

> As far as I can tell this option isn't usable for submodules which don't track master very closely. If you set depth 1, then submodule update can only ever succeed if the submodule commit you want is the latest master. Otherwise you get "fatal: reference is not a tree".

Additional comment: According to the same stackoverflow thread, Git 2.8 seems to have fixed this. My machine used to have Git 2.7.1... I've then upgraded it and still the `depth` failed.

```
Fetched in submodule path 'third-party/openFrameworks', but it did not contain 1145699df775d0df367af09f9f35b76eea560721. Direct fetching of that commit failed.
```

nebgnahz, 2016-05-03:
Alternative solution: Create a branch and git submodule will track that branch (http://stackoverflow.com/questions/9189575/git-submodule-tracking-latest)
```
[submodule "module-name"]
  path = module-path
  url = git@github.com:submodule/module.git
  branch  = branch
```

damellis, 2016-05-03:
Would a release work? If so, we could point at the openFrameworks release tag we want to use.

nebgnahz, 2016-05-04:
I am thinking of directly download the required version instead of using a submodule.
We are unlikely to change openFramework core and version tracking here seems unnecessary.

nebgnahz, 2016-05-23:
We are using of releases now.

closed 2016-05-23 by nebgnahz