damellis/ESP #179: Mistmatch in number of audio channels?

damellis, 2016-05-03:
The call to sound_stream_->setup() asks for two input channels, but then seems to treat the incoming data as if it only had one channel. Not sure what's going on.

nebgnahz, 2016-05-03:
The ofSoundStream API was confusing.

I couldn't configure sound_stream to use one channel. Two channels are not useful on Mac; and one channel is the case for Arduino electret sensor.

Also I failed to change the sampling frequency directly. The function failed to work if supplied a different frequency.

The solution I had was to perform channel combination and subsample in the ofSoundStream callback.

damellis, 2016-05-03:
Hmm... from my reading of the buffer format, the left and right audio samples alternate in the buffer, so we want to grab every other sample from the whole buffer rather than all the samples in the first half of the buffer. I made a corresponding change here: https://github.com/damellis/ESP/commit/5ec3a3c8671431da266d67b51b3e1088486b804e

Does that seem right to you? Or am I confused?

nebgnahz, 2016-05-03:
I think the previous code wasn't properly extracting the left channel. It got wrong probably during one iteration (software aging...).

Although I am still not certain about `buffer_size`: does it mean the total buffer size shared by 2 channels (each has buffer_size/2 data) or used by each channel (each has buffer_size data)?

damellis, 2016-05-03:
I believe buffer_size is the total size of the buffer, and each channel has buffer_size / nChannel elements in it.

closed 2016-05-04 by damellis