damellis/ESP #193: Xcode project can't find GRT/GRT.h

narner, 2016-05-07:
Hi there, 

First off, this looks like a really fantastic project! I'm looking forward to using it 👍 

I followed the instructions in the README for downloading the project and running the `./setup` command, but am getting the errors shown in the screen-shot below when running the Xcode project. I'm on OS X 10.11.4 and running Xcode 7.3. 

[image]

damellis, 2016-05-07:
I think this might be related to #178. Basically, we're referencing a particular commit of openFrameworks in our git submodule, but only git cloning the latest commit of openFrameworks in the setup script (because the entire oF repository is huge). Is your ESP/third-party/openFrameworks directory empty? If so, try running "git reset --hard" from that directory, then re-running the setup script from the top-level ESP directory.

nebgnahz, 2016-05-07:
Yes, the issue was related with the shallow clone. I have created a new pull request to address this: https://github.com/damellis/ESP/pull/194.

@damellis Can you review and merge it.

@narner You can give it a shot with `./setup minimal`.

Meanwhile, there is another pull request to homebrew so that you can install GRT with:
```
brew install grt
```

nebgnahz, 2016-05-07:
Fun fact, my current working directory for this project is 4.6G. Third party projects are 2.6G.

```
Bens-MacBook-Pro:smart-sensors benzh$ du -h -d 1
2.0G	./.git
 10M	./Arduino
 84K	./Processing
2.6G	./third-party
 30M	./Xcode
4.6G	.
```

With minimal installation, it's only around 1G.
```
Bens-MacBook-Pro:ESP benzh$ du -h -d 1
 57M	./.git
 10M	./Arduino
 84K	./Processing
914M	./third-party
 30M	./Xcode
1011M	.
```

narner, 2016-05-07:
@nebgnahz Yeah, cutting down the size of the project would definitely be helpful. Plus, I already have the GRT installed elsewhere on my system for an existing project.

narner, 2016-05-09:
@nebgnahz I saw you added the `./setup minimal` script. I pulled the most recent changes and gave the script a shot, but still got the same errors that I had originally.

nebgnahz, 2016-05-10:
This is probably due to a mistake in the script (see https://github.com/damellis/ESP/blob/master/setup#L27).

I've fixed it but am planning to test it tonight.

nebgnahz, 2016-05-10:
The current script should work.

narner, 2016-05-10:
@nebgnahz It looks like the script improved things, but I do still get two errors when I run `./setup minimal`: 

[image]

narner, 2016-05-10:
I saw you posted some new script changes (commit https://github.com/damellis/ESP/commit/1aec2db3fe16bd973d8e6bae85d0e654b405a2ec). That seemed to fix the OF issues, but I have a new error involving the GRT...should I close this issue and open a new one, or would you like me to attach a screenshot here of the GRT error?

nebgnahz, 2016-05-10:
I am glad it works. Have to fiddle around with `cp` and `mv` on Mac and decided to use `tar` which seems to work better.

Regarding the GRT error, please open another issue. That would help us tracking down different problems.

narner, 2016-05-11:
Sounds good, I'll close this one and open a new issue.

closed 2016-05-11 by narner