damellis/ESP #195: Change all user code to `.cpp` file

nebgnahz, 2016-05-08:
This avoids re-compilation.

Currently we are using `.h` file as user examples. Every small changes in `.h` file will affect every other file that has `#include` it, hence re-compilation of almost every single file. Using `.cpp` can sped up the process.

nebgnahz, 2016-05-17:
The analysis above doesn't hold here.

It turns out we are compiling 26 items every time (I am clueless why Xcode doesn't check modify time of each source file).

narner, 2016-05-17:
Having used Xcode for awhile; that does sound like "expected behavior", unfortunately.

nebgnahz, 2016-05-18:
That's a bummer! I've explored a bit more here and no success so far:

1. No luck of changing Xcode configuration to speed up.

1. OF Makefile helps eliminating re-compiling add-on `.cpp` files. But still compilation is slow:
```
$ touch src/ESP.h
$ time make -j8 Debug
...
make -j8 Debug  8.07s user 1.40s system 90% cpu 10.498 total
```

2. There exists some effort of improving openFramework (such as using CMake). I haven't seen them merged for a couple of good reasons...

nebgnahz, 2016-05-21:
Got CMake working in a separate branch
https://github.com/damellis/ESP/tree/cmake

Adding one line space in `ofApp.cpp` takes 5 seconds with CMake (in contrast, the original Make takes about 8 seconds). Additional speedup is possible since we are not linking against all openFramework libraries, a selective configuration would help.

Then again, refactoring our code to avoid extensive `#includes` and convert user program to `.cpp` is another option. The challenge here is that new changes should NOT break Xcode configuration!

damellis, 2016-05-21:
Would that mean users would have to install cmake separately? Or does it come with Xcode, etc?

nebgnahz, 2016-05-21:
Xcode would still work (so users can still use Xcode). This is more for us, who frequent compiles the code.

I think CMake doesn't come with Xcode, at least I recall using `brew install cmake`.

damellis, 2016-05-22:
Ah sorry, I didn't realize this was an alternative to Xcode compilation! (I thought it was changing the way the Xcode compilation worked.) So never mind...

nebgnahz, 2016-05-23:
After using CMake and moving implementations into .cpp file (avoid .h inlining), the compilation speed has been improved.

The major cause of the slow down comes from `ofMain.h` inclusion. We need it in AudioStream, so it's included in `istream.h`, `iostream.h`, `ESP.h` and `user_xx.cpp` file. I don't have a good solution to this.

Consider this a solved issue.

closed 2016-05-23 by nebgnahz