damellis/ESP #196: GRT/GRT.h file not found

narner, 2016-05-11:
After pulling commit 1aec2db3fe16bd973d8e6bae85d0e654b405a2ec and running `./setup minimal`, I get a build error in `istream.h` of `GRT/GRT.h file not found`.

[image]

nebgnahz, 2016-05-11:
tldr: use `brew install grt`.

Before today, I have GRT installed in `/usr/local/include/`; hence `GRT/GRT.h` can be found.

Then I removed it for testing, the same error you have occurs saying "GRT/GRT.h file not found". There is a Homebrew formula (https://github.com/Homebrew/homebrew-core/commit/d43a097c9f503410abb784fcf75bd3d784120977) so I used `brew install grt`. After a minute or so, the installation succeeded. GRT headers are installed in `/usr/local/include` and that's configured already in ESP Xcode project to search for. Now I am able to recompile this again.

If Homebrew doesn't work, then manually installing GRT seems to be the only option as of now.

nebgnahz, 2016-05-11:
You mentioned you have GRT installed somewhere already (in https://github.com/damellis/ESP/issues/193). Can you double check if it's installed in `/usr/local`, since this is the only _system_ path the Xcode project is search header for.

narner, 2016-05-11:
That makes sense. I just checked; and it looks as if it isn't installed there (I think I know why that is; a long tldr story). I'll try using `brew install grt` and see if that works.

narner, 2016-05-11:
That seemed to work, and the project builds now. Thanks! Looking forward to exploring this now :)

closed 2016-05-11 by narner