damellis/ESP #205: Command line prediction mode.

damellis, 2016-05-23:
w/o openFrameworks, once the model is trained.

damellis, 2016-09-10:
Or, rather, with openFrameworks but running headless. Except we need a way to load the model / ESP session.

nebgnahz, 2016-09-13:
So it works fine on Linux but fails miserably right now on Mac OS...

How to use the feature:

1. When building with CMake, you can use `cmake -DBUILD_HEADLESS=on ..`
which is equivalent of `#define HEADLESS` and the main file will be
compile as dictated.

2. When using Xcode or other IDE, users can directly edit `main.cpp`
file by adding the defined variable `HEADLESS`.

Works on Linux with `./ESP <session>`. On Mac OS, two troubles:

1. Xcode would add additional arguments. It's equivalent of running `/Users/benzh/repos/ESP/Xcode/ESP/bin/ESPDebug.app/Contents/MacOS/ESPDebug` `-NSDocumentRevisionsDebugMode` `YES`.

2. ofxDatGui::setAssetPath is magic... I failed to trace what happens there. So running on Mac OS will have infinite `loadFont` and `loadImage` errors.

nebgnahz, 2016-09-13:
For some reason, the command-line mode seems to be working on Mac OS as well. Do `open ESP.app --args <ESP-session-path>` will work.

Previously I was doing `ESP.app/Contents/MacOS/ESP <ESP-session-path>` and that causes the loadFont issue (understandably).

Still not sure how to make this run inside Xcode though.

damellis, 2016-09-13:
On OS X, we get lots of: `[ error ] ofTrueTypeFont: drawString(): font not allocated` messages. This is running: `./ESPDebug.app/Contents/MacOS/ESPDebug ~/Documents/Berkeley/ESP/Data/Accelerometer-Arduino101-ForehandBackhandServe/`

damellis, 2016-09-14:
Let's just #ifdef out the calls to create and draw the GUI in headless mode.