damellis/ESP #207: Compile GRT instead of using static library

nebgnahz, 2016-05-25:
We will need to make changes to GRT; plus adding it to `./setup` should be straightforward. Changing CMake is also straightforward. Need to evaluate Xcode...

nebgnahz, 2016-05-26:
Commit 1b5660403cbdf5b11a049406a14854cca8bdcb51 directly includes Filter module in ESP codebase instead of GRT. This avoids re-building GRT, thus no need to update `libgrt.dylib`.

Should this be the intended structure?

damellis, 2016-05-26:
Good question. To me, it was convenient because it didn't require re-building the GRT. But in the long run, we may want to send our additions upstream, which will be easier if they're in the GRT repository. How hard do you think it will be to compile the GRT?

nebgnahz, 2016-05-26:
It's fairly straightforward with CMake; the current CMake setup in GRT is easy to extend (see an example here: https://github.com/damellis/grt/commit/317b9ce568b5e54f6e992a93beb06201319bcd96, and there were more changes than it needed because of spaces/tabs issue).

I'm leaning toward a clean separation.

damellis, 2016-05-26:
Using CMake to compile build changes to the GRT seems reasonable, but it might be good to commit the updated .dylib to this repository so that Xcode users don't need to compile the GRT (and install CMake). What do you think?

damellis, 2016-05-27:
For reference, here's the command to run on the libgrt.dylib to set its install path correctly (corresponding to its use in the ESP Xcode project):

```
install_name_tool -id @executable_path/../Frameworks/libgrt.dylib libgrt.dylib
```

Maybe we want a custom "make install" action to instal the GRT dylib into the ESP project? Or just a script we can run?

nebgnahz, 2016-07-05:
There isn't a decision made on this issue. It looks like we are currently putting custom processing modules (MFCC, e.g.) in ESP instead of GRT. This works fine for now and there doesn't seem to be more custom processing modules at this stage.

The extensibility of GRT is being questioned and can only be resolved with a discussion with Nick. I am closing this issue now; we can reopen this if needed in the future.

closed 2016-07-05 by nebgnahz