damellis/ESP #209: Build under Linux.

damellis, 2016-05-26:


nebgnahz, 2016-05-26:
The build script is merged (a47f35a72e0cfbfbdfbb1773d7686817ef3c4e49) but bugs remain (https://github.com/damellis/ESP/issues/211).

nebgnahz, 2016-05-26:
Replicate my comment here:

Tested on Ubuntu 16.04
Compilation works. To run the audio example, you will need to invoke `jackd` separately. On my machine, I did jackd -d alsa --device hw:0 --rate 44100 --period 128. Learned from this thread: https://forum.openframeworks.cc/t/ofxsupercolliderserver-on-linux64/14623, but haven't investigated much.

nebgnahz, 2016-05-31:
With https://github.com/damellis/ESP/pull/218, we can now compile all examples under Linux. To run the audio, you still need:
`jackd -d alsa --device hw:0 --rate 44100 --period 128`

Documentation will be updated accordingly soon.

nebgnahz, 2016-06-01:
Commit 38662c82832b3baf06369ae686828350cdbf1bc2 and 0f38abab1941c4deb6fe0072a53d472c662a6b56 updated README file. So this is considered done.

closed 2016-06-01 by nebgnahz