damellis/ESP #215: Speaker identification.

damellis, 2016-05-27:
That is, figuring out who's talking. There are different versions of this, some of which only work on a fixed set of people, some of which work on arbitrary people; some require a fixed phrase, some don't; etc. 

Two related papers with what seems like a reasonable balance of implementation difficulty and performance:
https://www.ll.mit.edu/publications/journal/pdf/vol08_no2/8.2.4.speakerrecognition.pdf
http://www.cs.toronto.edu/~frank/csc401/readings/ReynoldsRose.pdf

nebgnahz, 2016-07-10:
I am hitting the performance issue now.

A few reasons:
1. GMM is known to perform badly when the dimension increases; GRT also warns about it (see the Disadvantage part of http://www.nickgillian.com/wiki/pmwiki.php/GRT/GMMClassifier).
2. The process of training is essentially [running raw data through the pipeline](https://github.com/nickgillian/grt/blob/1d99c9c3b2e988f38a57f08b7cb8d3d4221db2a0/GRT/CoreModules/GestureRecognitionPipeline.cpp#L396). For audio data, it's a lot of loops. And in every iteration, the whole pipeline is executed.
3. Currently our MFCC is a straightforward implementation that doesn't have much optimization.
4. For a reasonable detection rate, both the data size as well as the model complexity seems necessary (haven't done more evaluation here).

I can work to improve 3. But mainly 2 is over-killing. Loop evaluation vs. assembly Matrix multiplication.

Alternatively we can implement a off-line training program and enable the GUI to load a trained model. But this won't allow easy tinkering because any changes made to the dataset will require long-time waiting.

Quick note: running a Matlab MFCC+GMM implementation takes about 30 seconds to train over 30 males and 30 females audios (each are about one sentence, 3-4 seconds) with GMM degree 32. For GRT's implementation, I've waited for more than 10 minutes.

nebgnahz, 2016-07-13:
Tuning various parameter doesn't help much reducing the training time. We are hitting the performance wall before the GMM step. Since most of the time was spent in get MFCC features, I am working on improving this module.

I did an initial attempt to speed up MFCC feature extraction. Using BLAS `cblas_dgemv` for the triband filter, I got 3x improvements. I am planning to also optimize the DCT operation and see if how much this helps.

Note that the work is currently kept in [a separate repository](https://github.com/nebgnahz/speaker-identification/commit/0d832f4e2db4650721a34b8e95157f14ee3823f8).

nebgnahz, 2016-07-13:
I've managed to reduce the MFCC feature extraction significantly. Running the test 100 iterations, the runtime has reduced from 2 seconds to 0.6 seconds (see my last comment) and now 0.12 seconds (roughly 16x improvement).

This now makes the bottleneck being the training phase. But I was able to complete a few training configurations (loading 10 audio files, use 5 cepstral coefficients and 5 mixture models, see https://github.com/nebgnahz/speaker-identification/blob/master/speaker-recognition.cpp). Unfortunately, the results are completely off (for some reason all are classified as male speaking). I attempted to increase the number of cepstral coefficients and the mixture model but then the training is running for more than 10 mins.

I will incorporate the MFCC changes to ESP and check the feature computed lively. The current correctness is checked against the Matlab reference implementation (see https://github.com/nebgnahz/speaker-identification/blob/master/mfcc-test.cc#L103).

damellis, 2016-07-13:
Thanks for the updates! Let me know if there's anything I can do to help with this.

nebgnahz, 2016-07-15:
The latest update on the progress is:
I am trying to use a simple voice level detector to filter silence and background noise (non-voice signals) but GRT's pipeline doesn't seem to be respecting `featureDataReady` flag. See https://github.com/nickgillian/grt/issues/75.

damellis, 2016-07-15:
Hmmm... maybe something we should fix and submit a pull request for?

nebgnahz, 2016-07-19:
I've prepared the [PR](https://github.com/nickgillian/grt/pull/76). With this, the total execution time (loading + training + predicting) is on the order of a few minutes.

I have ran a few configurations and measured both the accuracy (a rough estimate, correct/all) and the execution time. 

|    MFCC |     GMM | Training | Accuracy | Time (seconds) |
| :-----: | :-----: |  :-----: |  :-----: |        :-----: |
|       6 |      16 |       10 | 0.880013 |        30.4071 |
|       7 |      16 |       10 | 0.865261 |        34.2501 |
|       8 |      16 |       10 | 0.790613 |        40.4655 |
|       9 |      16 |       10 | 0.923199 |        47.9614 |
|      10 |      16 |       10 |     -nan |        15.5639 |
|      11 |      16 |       10 | 0.984345 |        62.9507 |
|      12 |      16 |       10 | 0.808999 |        71.8248 |
|      12 |      12 |       10 | 0.777819 |        54.8725 |
|      12 |      16 |       10 | 0.887068 |        71.8398 |
|      12 |      20 |       10 | 0.692481 |        90.6852 |
|      12 |      24 |       10 |  0.83084 |        106.832 |
|      12 |      28 |       10 | 0.871675 |        123.666 |
|      12 |      32 |       10 | 0.902247 |        141.975 |
|      12 |      16 |       10 | 0.894123 |        72.5912 |
|      12 |      16 |       15 |  0.98106 |        93.6182 |
|      12 |      16 |       20 | 0.896047 |         110.05 |
|      12 |      16 |       25 | 0.891205 |        125.245 |
|      12 |      16 |       30 | 0.933173 |        139.044 |

It should be noted that given the nature of GMM (EM), different runs with the same configuration might yield different accuracy and execution time because of the randomized initialization.

P.S. The machine's configuration is: Intel(R) Xeon(R) CPU E5-1620 v2 @ 3.70GHz, 32G RAM.

nebgnahz, 2016-07-27:
GRT's GMM behaves quite randomly: different runs will have different result, some succeeded in training, some failed.

I am in the progress of working on this command line program that would facilitate debugging. It would be less interactive than ESP but easier to do trial-and-test (since no need to wait for the GUI to startup, or manually load data, train the pipeline).

See: https://github.com/damellis/ESP/blob/master/Xcode/ESP/src/cli.cpp#L22

nebgnahz, 2016-07-27:
It looks like working...

[image]
[image]

nebgnahz, 2016-07-28:
The reason that file loading fails is because of round-off errors (the accumulated errors makes the gauss producing +inf).

This is fixed now (https://github.com/damellis/grt/commit/4e774355afdce7c5171e81c875f60acce28bf950).

nebgnahz, 2016-08-10:
I think we can declare this done as the following commits are made:

1. https://github.com/damellis/grt/commit/f97b01c20edac80990118626c059251d8d14610a
During prediction, the prediction isn't successful if the `featureDataReady` flag hasn't been set.

2. https://github.com/damellis/ESP/commit/a106f2e1e4991d4132026d8a7d9a1c8a4c94d337
Move `user_audio.cpp` as `user_speaker.cpp` and add tunable parameter `confidence`.

closed 2016-08-10 by nebgnahz

damellis, 2016-08-10:
Sounds good!