damellis/ESP #246: Record events / keystrokes / interactions with system.

damellis, 2016-06-30:
For evaluation purposes. So we don't have to watch the video / screen capture of the user study.

damellis, 2016-08-02:
* edits to training data
* edits to test data
* training model
* switching tabs

nebgnahz, 2016-08-10:
Create a log file and save events (with timestamps and context).

damellis, 2016-08-11:
Thanks for doing this so quickly! Overall, it seems to be working well, but I had a couple of specific concerns:

1. We should probably move the log files out of the binary / build directory. Otherwise they get deleted if, for example, you do a "Clean" in Xcode. Maybe they should go to something like "~/.esp/" (Linux) / "~/Library/Application Support/ESP/" (Mac) / similar on Windows. Thoughts? Do you have time to take a shot at this?
2. It looks like logging to a file means that our previous log messages no longer show up in the console. This might make it easier to miss problems. How hard do you think it would be to log to both the log file and the console? (We might want to set the file log, and these interaction traces, to "verbose" with the console log at "notice".)

Also, do you think we need the logging of individual keystrokes? It seems to add a lot of noise to the logs. (Apologies if I suggested including them in our conversation.)

nebgnahz, 2016-08-11:
Comment 1 is partially covered by https://github.com/damellis/ESP/commit/df83b8340b136bf5e3f61194ea7232590e827262. I have no idea what to do for Windows (also don't have a Windows now for testing).

Regarding comment 2, it's hard as we are using `ofLog`. It only has a single `ofBaseLoggerChannel` (https://github.com/openframeworks/openFrameworks/blob/master/libs/openFrameworks/utils/ofLog.h#L477). You may toggle between the two modes, but that will incur [file open](https://github.com/openframeworks/openFrameworks/blob/master/libs/openFrameworks/utils/ofLog.cpp#L53) every time. Alternatively, we can take ownership of the `ofFileLoggerChannel` and set it manually, but I am worried the potentially-introduced bug because of it.
The reason I used `ofLog` in the first place is that I believe it should have been well tested. Given the limited time we have to prepare for the workshop, rolling our own file management doesn't seem wise. But now the downside is, there is only one place that `ofLog` can go. We can, temporarily, add additional `std::cout` as it's for sure going to the console and portable/well-tested.

Individual keystrokes are annoying if you are looking at the log file as is manually. Simple tools, like `grep -v "keyPressed" *.txt | grep -v "keyReleased"`, would work. Generally speaking, I am not very concerned about post-processing of the log files. But if you think they contribute little, removing them is fine.

damellis, 2016-08-11:
I added code to use %APPDATA%/ESP on Windows: https://github.com/damellis/ESP/commit/16ce0720dfb255d3d13d88083bd6419d23cbfa7f

nebgnahz, 2016-08-12:
This is also tested on Linux (Ubuntu) as well. Working fine.

damellis, 2016-08-12:
Closing as this seems to be working. I agree, it's not hard to filter out the key-pressed and key-released messages.

closed 2016-08-12 by damellis