damellis/ESP #250: Need to tell runUnitTests how to find libgrt.dylib (OS X).

damellis, 2016-07-05:
I get the following error when running runUnitTests:

```
dyld: Library not loaded: @executable_path/../Frameworks/libgrt.dylib
  Referenced from: /Users/mellis/Source/ESP/./runUnitTests
  Reason: image not found
Trace/BPT trap: 5
```

It looks like this can be fixed by telling runUnitTests where to find libgrt.dylib:

```
install_name_tool -change @executable_path/../Frameworks/libgrt.dylib Xcode/ESP/libgrt.dylib runUnitTests
```

But I'm not sure how to tell cmake to insert this into the Makefile. @nebgnahz: any ideas?

nebgnahz, 2016-07-05:
Fixed.

nebgnahz, 2016-07-05:
The idea is to run a custom command (`install_name_tool`) after build (see `POST_BUILD` argument).

Although I am not sure how the path `../Framework/libgrt.dylib` is added to the build process...

closed 2016-07-05 by nebgnahz

damellis, 2016-07-05:
Thanks! It seems to work now. The ../Framework/libgrt.dylib is something I've been doing manually, although I recently added it to the https://github.com/damellis/ESP/blob/master/update-xcode-grt.sh script. It's needed for the ESP executables built with Xcode to find the grt.