damellis/ESP #252: TrainingDataManager::load() needs to process loaded data.

damellis, 2016-07-05:
Currently, none of the members of TrainingDataManger are updated when new training data is loaded into the underlying GRT::TimeSeriesClassificationData. That makes the manager's state inconsistent and assertions start failing.

nebgnahz, 2016-07-06:
This should have been fixed by:
https://github.com/damellis/ESP/commit/b000a4c8fc8be91a7ef1fe6864a7d4987a871b28, https://github.com/damellis/ESP/commit/088d17a3272918af72b47d342b9c8e1ac60aee2e, https://github.com/damellis/ESP/commit/69de1322c43080fb4f4e9207c72a4318a972447f.

Unit test is passed but it might not cover all corner cases.
Please close this if it's actually fixed.

damellis, 2016-07-06:
It seems like there are still some issues. Specifically, our ofApp assumes that there are at least 9 classes but loading a saved TimeSeriesClassificationData only counts the number of classes that actually have samples associated with them. This leads to the assertion failure below. Maybe load() should use the larger of the current value of num_classes and data_.getNumClasses()?

```
#4	0x00000001000bb09e in TrainingDataManager::getNumSampleForLabel(unsigned int) at /Users/mellis/Source/ESP/Xcode/ESP/src/training-data-manager.cpp:100
#5	0x0000000100024128 in ofApp::loadTrainingData() at /Users/mellis/Source/ESP/Xcode/ESP/src/ofApp.cpp:1263
#6	0x00000001000250ac in ofApp::keyPressed(int) at /Users/mellis/Source/ESP/Xcode/ESP/src/ofApp.cpp:1352
#7	0x000000010002c211 in ofBaseApp::keyPressed(ofKeyEventArgs&) at /Users/mellis/Source/ESP/Xcode/ESP/../../third-party/openFrameworks/libs/openFrameworks/app/ofBaseApp.h:80
#8	0x000000010054462f in std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)::operator()(void const*, ofKeyEventArgs&) const [inlined] at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworksCompiled/project/osx/../../../../libs/openFrameworks/events/ofEvent.h:237
#9	0x000000010054460d in decltype(std::__1::forward<ofBaseApp>(fp)(std::__1::forward<void const*, ofKeyEventArgs&>(fp0))) std::__1::__invoke<std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)&, void const*, ofKeyEventArgs&>(ofBaseApp&&, void const*&&, ofKeyEventArgs&&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__functional_base:416
#10	0x000000010054460d in bool std::__1::__invoke_void_return_wrapper<bool>::__call<std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)&, void const*, ofKeyEventArgs&>(std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)&&&, void const*&&, ofKeyEventArgs&&&) [inlined] at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__functional_base:437
#11	0x000000010054460d in std::__1::__function::__func<std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&), std::__1::allocator<std::__1::shared_ptr<of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex> > ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::make_function<ofBaseApp>(ofBaseApp*, void (ofBaseApp::*)(ofKeyEventArgs&), int)::'lambda'(void const*, ofKeyEventArgs&)>, bool (void const*, ofKeyEventArgs&)>::operator()(void const*&&, ofKeyEventArgs&) at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:1437
#12	0x0000000100523e81 in std::__1::function<bool (void const*, ofKeyEventArgs&)>::operator()(void const*, ofKeyEventArgs&) const at /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/functional:1817
#13	0x0000000100523cb2 in of::priv::Function<ofKeyEventArgs, std::__1::recursive_mutex>::notify(void const*, ofKeyEventArgs&) [inlined] at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworksCompiled/project/osx/../../../../libs/openFrameworks/events/ofEvent.h:136
#14	0x0000000100523c93 in ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>::notify(void const*, ofKeyEventArgs&) at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworksCompiled/project/osx/../../../../libs/openFrameworks/events/ofEvent.h:303
#15	0x0000000100522c49 in void ofNotifyEvent<ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>, ofKeyEventArgs>(ofEvent<ofKeyEventArgs, std::__1::recursive_mutex>&, ofKeyEventArgs&) [inlined] at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworksCompiled/project/osx/../../../../libs/openFrameworks/events/ofEventUtils.h:222
#16	0x0000000100522c3b in ofCoreEvents::notifyKeyPressed(int, int, int, int) at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworks/events/ofEvents.cpp:251
#17	0x00000001002ea3d1 in -[GLFWContentView keyDown:] ()
#18	0x00007fff934dfb15 in -[NSWindow _reallySendEvent:isDelayedEvent:] ()
#19	0x00007fff92f1e539 in -[NSWindow sendEvent:] ()
#20	0x00007fff92e9f16a in -[NSApplication sendEvent:] ()
#21	0x00000001002eaa28 in -[GLFWApplication sendEvent:] ()
#22	0x00000001002eb924 in _glfwPlatformPollEvents ()
#23	0x0000000100543755 in ofMainLoop::pollEvents() [inlined] at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworks/app/ofMainLoop.cpp:148
[... cut ...]
#25	0x000000010051f125 in ofRunMainLoop() [inlined] at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworks/app/ofAppRunner.cpp:173
#26	0x000000010051f118 in ofRunApp(std::__1::shared_ptr<ofBaseApp>) at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworks/app/ofAppRunner.cpp:159
#27	0x000000010051f09d in ofRunApp(ofBaseApp*) at /Users/mellis/Source/ESP/third-party/openFrameworks/libs/openFrameworks/app/ofAppRunner.cpp:153
#28	0x000000010000195a in main at /Users/mellis/Source/ESP/Xcode/ESP/src/main.cpp:18
#29	0x00000001000017b4 in start ()
```

nebgnahz, 2016-07-06:
Good catch. Fixed (https://github.com/damellis/ESP/commit/5a2b39937adfcdc67f1d5ac44a3e5de205f89157).

damellis, 2016-07-08:
With your commits and https://github.com/damellis/ESP/commit/c21bd47ad36112991c258df4b4bd719198273310, this seems to be fixed.

closed 2016-07-08 by damellis