damellis/ESP #254: Behavior of save/load (serialization/deserialization)

nebgnahz, 2016-07-06:
Issue (https://github.com/damellis/ESP/issues/252) raises a question about what we should do regarding saving/loading. Currently GRT provides save/load from/to files and uses a custom format. And since we are adding additional functionality (especially the per-sample naming), how should we integrate this with GRT?

Below are a few possibilities for discussion:

**Proposal 1:**
Use multiple files and a custom format. GRT has its own files for saving/loading. We add our own files in parallel to GRT's and ask the `TrainingDataManager` to handle combining those. This could be made similar to the directory-based save-all approach (https://github.com/damellis/ESP/issues/150). We are already using some custom format for the tunable parameters. 

**Proposal 2:**
Use multiple files and some well-known format. Formats that I can think of include JSON, YAML, Protocol Buffer, etc. This will introduce additional dependencies (it's a better idea to use some serialization library than roll your own).

**Proposal 3:**
Use a single `*.esp` file. This requires changing GRT so that it doesn't save/load from file but instead serialize/deserialize from a `string` of `stringstream`. In this way, we can combine all existing modules and save them simultaneously.

narner, 2016-07-06:
Hey @nebgnahz I noticed this issue as I'm "watching" the ESP project...the GRT's [ClassificationData](http://www.nickgillian.com/wiki/pmwiki.php/GRT/ClassificationData) structure can also save/load files that are in .CSV format (see line 68 on this file: https://github.com/nickgillian/grt/blob/master/examples/DatastructuresExamples/ClassificationDataExample/ClassificationDataExample.cpp). Not sure if that would be helpful for your Proposal 2 section; just thought it would be helpful in case you hadn't noticed that.

nebgnahz, 2016-07-06:
@narner Thanks for the pointer and suggestion. I didn't mention CSV file because CSV is good for data (`MatrixDouble` internally) but not for **meta-data**. Specifically, here we are a bit concerned about the names of each labelled class and even each individual collected samples.

In our own fork of GRT, we've added custom class name saving/loading ourselves (https://github.com/damellis/grt/commit/4177f2a082b386878b765d449c40806efdbd4877). There we simply followed the format GRT was using. Now since we are gradually [adding support for each individual samples](https://github.com/damellis/ESP/blob/master/Xcode/ESP/src/training-data-manager.h#L30), I bring up the issue so that the decision would be helpful for future extensions.

narner, 2016-07-06:
@nebgnahz Gotchya, all makes sense!

damellis, 2016-07-06:
A few thoughts:

1. Fixing saving and loading is much more important than per-sample names. There are lots of questions about how the latter will work, and they're not necessarily super useful. So it would be good to fix loading (without per-sample names) now, while we decide on an overall strategy.

2. For the longer-term, I'd think we should continue to use the GRT serialization format and switch to a directory-based (or file prefix-based) approach to saving and loading so the user doesn't have to deal with individual files. Porting the GRT to a different / standard serialization scheme / format is something that probably needs to happen, but I think it should happen in the GRT itself (https://github.com/nickgillian/grt/issues/66). Similarly with modifying the GRT to use generic streams instead of file streams; this is something I started working on in my AVR / ARM port (https://github.com/damellis/grt/tree/avr) but it's a bigger task that we probably want to avoid for ESP.

3. Given the brittleness of the GRT's current serialization format, it seems like we should either use it completely unchanged, or change it as much as we need to. There's not really any benefit to making some changes but not others, since any changes will (I think) break compatibility with the standard GRT. Specifically, it seems like we should either remove class names from our GRT save files (to restore compatibility) or add per-sample names to them (if we're not worrying about compatibility). If the former, I'm generally agnostic as to which file format we use for additional ESP metadata beyond that stored by the standard GRT.

@nebgnahz: what do you think?

nebgnahz, 2016-07-06:
I partially agree with 1 (hence the fix https://github.com/damellis/ESP/commit/b000a4c8fc8be91a7ef1fe6864a7d4987a871b28). The part that I don't agree is the "not necessarily super useful". If we are giving the ability to name individual samples, the inconsistent behavior of saving "A" but getting "B" after loading doesn't seem to be appropriate.

For 2, I agree it should be GRT's work. That's why I put a similar approach as proposal 3 instead 1. And in proposal 1/2, the directory-based seems fine.

Regarding 3, personally I feel it was a bug in GRT instead of compatibility issue of not loading class name. With that said, we should really create a PR for it. Adding per-sample naming is breaking the compatibility. The question is how important that feature is to GRT. It does seem important to us (heard a couple times when users are trying to mix expert training data and their own training data).

damellis, 2016-07-26:
It seems that we settled on sticking with the existing GRT formats for now, plus adding our own files for tunable parameters. I'm closing this for now. Please re-open if necessary.

closed 2016-07-26 by damellis