damellis/ESP #263: Merge Plotter and InteractiveTimeSeriesPlot classes.

damellis, 2016-07-08:
Currently, we have two classes for plotting that allow the user to select a range of the plot. It would be good to unify these.

damellis, 2016-07-08:
One difference is that the ofxGrtTimeseriesPlot assumes a fixed data buffer length (i.e. fixed number of x-axis data points) while Plotter allows plotting of data of arbitrary lengths.

nebgnahz, 2016-07-08:
Apologies that I should have documented these two classes: why they exist and what they differ.

The fixed size buffer is the major one. `Plotter` uses `GRT::MatrixDouble data_` as the data store while `InteractiveTimeSeriesPlot` relies on the base class `ofxGrtTimeseriesPlot` which uses a circular buffer (hence the fixed size buffer).

The other reason to have `InteractiveTimeSeriesPlot` was to make as minimal changes as possible to `ofApp` but add the functionality of interactiveness. This leads to a subclass implementation.

Besides the data storage, renaming is only supported in `Plotter`; also the subtitle.

nebgnahz, 2016-07-16:
Use `Plotter` for non-live data (such as training data).
Use `InteractiveTimeSeriesPlot` for live data (utilizing the underlying circular buffer store).

closed 2016-07-16 by nebgnahz