damellis/ESP #279: Sync GRT HEAD?

nebgnahz, 2016-07-19:
I am under the impression that we take a snapshot of GRT and use it for most of the development.

However, given the [PR](https://github.com/nickgillian/grt/pull/76) and other changes that has been made in GRT, should we consider syncing our GRT?

There are some type changes that we will need to address if we sync. E.g. `MatrixDouble` is no longer available and it's now `MatrixFloat`. If you dig more about what's the float, you will see [`typedef double Float`](https://github.com/nickgillian/grt/blob/master/GRT/Util/GRTTypedefs.h#L51).

damellis, 2016-07-19:
I think it's probably a good idea. Any thoughts on how we should go about it?

nebgnahz, 2016-07-19:
I think it would be helpful to still use your forks: syncing that to the master and then update our submodule. Then fix any compilation and API changes (this could take a while...).