damellis/ESP #315: Update grt library on Windows and Linux.

damellis, 2016-08-10:


damellis, 2016-08-12:
Updated the Windows libraries here: https://github.com/damellis/grt/releases and here: https://github.com/damellis/ESP/wiki/Windows-Setup-Guide (to https://github.com/damellis/grt/tree/e1cd0b5).

nebgnahz, 2016-08-12:
I augmented the CMakeLists for Linux such that if GRT is not installed on the system, we use `Xcode/ESP/libgrt.a`. See https://github.com/damellis/ESP/commit/5df27ff56b079a623d5d1d1a5ddacdfe8e07d029.

And I have built the latest version of GRT here:
[Debug -- 72.5 MB](https://dl.dropboxusercontent.com/u/4388609/libgrt-debug.a)
[Release -- 9.8 MB](https://dl.dropboxusercontent.com/u/4388609/libgrt-release.a)

closed 2016-08-12 by damellis