damellis/ESP #323: Build binary releases of ESP examples on Linux.

damellis, 2016-08-12:
@nebgnahz do you have time to do this today? (It's probably not as important as on Mac and Windows, since the compilation pre-requisites aren't as big / slow to download but it would still be nice.) The gesture recognition example is the most important. The Touche, speaker recognition, and color sensor examples would also be nice.

I'll do the Mac and Windows ones.

We should also link these from the wiki; probably from the documentation for the example.

nebgnahz, 2016-08-12:
It's a bit harder than I thought: packaging all the shared libraries in a single executable...

Here is a list of dependencies for the final executable:

```
$ ldd ESP
	linux-vdso.so.1 =>  (0x00007ffe7a297000)
	libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f993bcdc000)
	libfreeimage.so.3 => /usr/lib/x86_64-linux-gnu/libfreeimage.so.3 (0x00007f993ba29000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f993b77e000)
	libGLEW.so.1.13 => /usr/lib/x86_64-linux-gnu/libGLEW.so.1.13 (0x00007f993b4fb000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f993b1ea000)
	libgstapp-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstapp-1.0.so.0 (0x00007f993afdb000)
	libgstbase-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstbase-1.0.so.0 (0x00007f993ad77000)
	libgstreamer-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0 (0x00007f993aa4d000)
	libgstvideo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libgstvideo-1.0.so.0 (0x00007f993a7c7000)
	libcrypto.so.1.0.0 => /lib/x86_64-linux-gnu/libcrypto.so.1.0.0 (0x00007f993a36c000)
	libssl.so.1.0.0 => /lib/x86_64-linux-gnu/libssl.so.1.0.0 (0x00007f993a103000)
	librtaudio.so.4v5 => /usr/lib/x86_64-linux-gnu/librtaudio.so.4v5 (0x00007f9939ee8000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f9939ca5000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f9939a52000)
	libopenal.so.1 => /usr/lib/x86_64-linux-gnu/libopenal.so.1 (0x00007f99397e1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f99395c4000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f9938f79000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f9938c3e000)
	libGL.so.1 => /usr/lib/nvidia-367/libGL.so.1 (0x00007f99389af000)
	libXxf86vm.so.1 => /usr/lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007f99387a9000)
	libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f993859d000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f993838d000)
	libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007f9938183000)
	libsndfile.so.1 => /usr/lib/x86_64-linux-gnu/libsndfile.so.1 (0x00007f9937f19000)
	libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007f9937d01000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007f9937afd000)
	libblas.so.3 => /usr/lib/libblas.so.3 (0x00007f993788d000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f993750b000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9937202000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9936feb000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9936c22000)
	libpixman-1.so.0 => /usr/lib/x86_64-linux-gnu/libpixman-1.so.0 (0x00007f993697a000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f9936754000)
	libxcb-shm.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007f9936550000)
	libxcb-render.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007f9936346000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f9936123000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f9935f19000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f9935d07000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f9935aec000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f99358e4000)
	libjxrglue.so.0 => /usr/lib/x86_64-linux-gnu/libjxrglue.so.0 (0x00007f99356c3000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f9935469000)
	libopenjp2.so.7 => /usr/lib/x86_64-linux-gnu/libopenjp2.so.7 (0x00007f9935231000)
	libraw.so.15 => /usr/lib/x86_64-linux-gnu/libraw.so.15 (0x00007f9934f71000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f9934cfd000)
	libwebpmux.so.1 => /usr/lib/x86_64-linux-gnu/libwebpmux.so.1 (0x00007f9934af6000)
	libwebp.so.5 => /usr/lib/x86_64-linux-gnu/libwebp.so.5 (0x00007f993489a000)
	libIlmImf-2_2.so.22 => /usr/lib/x86_64-linux-gnu/libIlmImf-2_2.so.22 (0x00007f99343cb000)
	libHalf.so.12 => /usr/lib/x86_64-linux-gnu/libHalf.so.12 (0x00007f9934188000)
	libIex-2_2.so.12 => /usr/lib/x86_64-linux-gnu/libIex-2_2.so.12 (0x00007f9933f6a000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f9933cf9000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f9933af5000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f99338f0000)
	liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007f9933670000)
	libasound.so.2 => /usr/lib/x86_64-linux-gnu/libasound.so.2 (0x00007f993336f000)
	libjack.so.0 => /usr/lib/x86_64-linux-gnu/libjack.so.0 (0x00007f9933126000)
	libpulse-simple.so.0 => /usr/lib/x86_64-linux-gnu/libpulse-simple.so.0 (0x00007f9932f21000)
	libpulse.so.0 => /usr/lib/x86_64-linux-gnu/libpulse.so.0 (0x00007f9932cd1000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f9932aa8000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f99328a0000)
	/lib64/ld-linux-x86-64.so.2 (0x00005563e25ba000)
	libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007f99325ea000)
	libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007f99323dd000)
[... 15 lines cut ...]
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f992fe5b000)
[... cut ...]
	libwrap.so.0 => /lib/x86_64-linux-gnu/libwrap.so.0 (0x00007f992dbda000)
	libasyncns.so.0 => /usr/lib/x86_64-linux-gnu/libasyncns.so.0 (0x00007f992d9d4000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f992d7ae000)
	libdatrie.so.1 => /usr/lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f992d5a6000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f992d2c5000)
	libnsl.so.1 => /lib/x86_64-linux-gnu/libnsl.so.1 (0x00007f992d0ab000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f992ce97000)
```

damellis, 2016-08-12:
Hmm… it might still be worth doing for the gesture recognition example, at least, if possible.

damellis, 2016-08-12:
BTW, I created a GitHub release here that you should be able to add any Linux binaries to: https://github.com/damellis/ESP/releases

nebgnahz, 2016-08-13:
I've added built binaries. Note these are built for UNIX-GNU ABI. Executing them on different ABI-based machines will cause illegal instructions. I am still learning the difference here and I am not confident if these would work out of box for any Linux machine.

Here is the ELF header:

```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 03 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - GNU
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x452540
  Start of program headers:          64 (bytes into file)
  Start of section headers:          9203048 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         10
  Size of section headers:           64 (bytes)
  Number of section headers:         42
  Section header string table index: 39
```

## Tips:

Using this command to copy all dependencies to `libs` folder:
```
cp `ldd ESP | grep -o '\W/[^ ]*'` libs
```

This [blog](https://h3manth.com/content/copying-shared-library-dependencies) has a script that's useful. AppImage seems to be a tool for this (I haven't tried); the script at the discussion on HN (https://news.ycombinator.com/item?id=11187198) seems to be sharing the same approach with the blog.

nebgnahz, 2016-08-13:
I have taken another look at AppImage, it could be helpful but 1) it's not easy to use directly 2) there is not enough time to get it done

probonopd, 2016-08-13:
Let me know what needs to be done and how I can help with the AppImage. I am the AppImage lead developer.

damellis, 2016-08-13:
Thanks @nebgnahz! We'll try the binaries and if they don't work, at least Linux users tend to have lots of experience building things from source... :)