damellis/ESP #347: Only accept keyboard and mouse input in relevant tab.

damellis, 2016-08-16:
I believe there are many keyboard and mouse commands that can be activated even when the relevant tab isn't open. We should restrict these commands to their corresponding tabs.

nebgnahz, 2016-08-22:
It's more than tabs. There are also different states associated with each tab: relabeling or renaming. A clean way might be defining the application state and model the behavior as a state machine.

nebgnahz, 2016-08-25:
I made a series of commits that are trying to address this issue. It seems to work fine, but more testing are needed. @damellis can you give it a shot?

damellis, 2016-08-26:
It looks like pressing '1', '2', etc. to collect calibration samples isn't working.

If we're going to remove the C/P/A/T/R keyboard shortcuts for switching tabs, we should probably also remove the brackets in the tab names (e.g. "[C]alibration" should become just "Calibration").

It looks like part of a training sample (for trimming) can be selected, even when in another tab besides training. Same with part of the overview on the analysis tab.

Also, the training data menu items (e.g. "delete", "rename", etc.) can be clicked when in another tab.

nebgnahz, 2016-08-26:
I've fixed calibration and the C/P/A/T/R. About the interference among tabs, I will have to look into it later.

nebgnahz, 2016-08-26:
I've fixed clicking the button in tabs issue. I can fix the selection as well, but since it is not directly involved with the `AppState`, I am thinking about holding it off.

This raises a bigger issue: `draw` function only controls drawing or not, the listeners (added by `ofAddListener`) are still working and able to capture UI events. And these listeners doesn't respect the layering or visibility. I see two approaches: 1, somehow explicitly disable them, as I did in the commit https://github.com/damellis/ESP/commit/3f0aec010960fc2535079fed4d30504c748b9a91; 2, leave them as is but try to guard against their influences by using the AppState, that is, the user might still observe some selection, but those selection won't take effect unless it obeys a certain rule.

P.S. We should start a new issue about it.

nebgnahz, 2016-08-28:
A follow-up comment on the workflow: when making large changes to the UI handling, it's difficult (and sometimes even impossible) to test every condition. I think we can follow a certain commit message format (see below) so that when looking at each changes, or when looking back at what happens, it will be easier to determine which commit is likely to introduce bugs.

- `fix(xxx): <message>` is usually a small fix that only involves changing one module (best if associated with an issue describing the problem first), thus testing that part would be sufficient. Small refactoring can be fixes.
- `feat(xxx): <message>` is a feature implementation, although there could be stand-alone features, usually this indicates a large commit and primarily focused on `xxx` and might have effect on other parts. We leave such judgement to the commit message.
- `refactor(xxx): <message>` is large changes that doesn't provide new features but could introduce bugs because of the size of the commit.

For feature and refactoring, we could consider using PR to avoid master branch failure. Inside that PR, we can explicitly say what testing have been done and what might be missing.

These are adapted from [Angular commit message template](https://github.com/angular/angular.js/blob/master/CONTRIBUTING.md#commit-message-format).

damellis, 2016-08-30:
Good enough. If there are remaining problems we'll open new issues.

closed 2016-08-30 by damellis