damellis/ESP #353: Disable event listeners for GUI elements on non-active tabs.

damellis, 2016-08-26:
This could involve explicitly disabling and re-enabling the listeners when the user switches tab, or we could have each event listener check the application state internally.

damellis, 2016-08-29:
This still seems to be a problem, despite: https://github.com/damellis/ESP/commit/3f0aec010960fc2535079fed4d30504c748b9a91

damellis, 2016-08-29:
This is probably the most important thing on the list, since accidentally clicking the buttons on the training data tab can delete data, etc.

nebgnahz, 2016-08-30:
I think the issue is here: https://github.com/braitsch/ofxDatGui/blob/master/src/ofxDatGui.cpp#L846

Even if we explicitly set the GUI elements to be disabled, ofXDatGui checks for mouse event to set the active element. And inside `ofxDatGui::focus()`, the `mEnabled` is set to be true.

Setting elements to be invisible would do the trick.

commit 27dd7cf2 (2016-08-30): fix(gui): calling setVisible instead of setEnabled

ofxDatGui would enable elements upon mouse clicking, so the previous
`setEnabled` doesn't work. Setting elements to be invisible would
disable ofxDatGui handling mouse/keyboard events.

Closes #353
closed 2016-08-30 by nebgnahz