damellis/ESP #367: Use separate pipeline for extraction of features from training samples.

damellis, 2016-09-09:
Currently, when in "feature view" (i.e. displaying live data and training samples as features), the features are dynamically calculated from the training samples when navigating between training samples. This uses the live pipeline, which creates glitches in the plot of the features of the live data. We should probably clone the pipeline and use the clone from calculating features of the training samples (and reset it before use).

damellis, 2016-09-23:
@nebgnahz this was fixed by https://github.com/damellis/ESP/commit/d99dbf766a4c5bbb04fc6c67b9b2d0e403be03b7 right?

nebgnahz, 2016-09-23:
I don't think so. It was planned but not fixed.

damellis, 2016-09-23:
Sorry, my mistake! I was confusing this with https://github.com/damellis/ESP/issues/337