damellis/ESP #369: Compilation and running on CHIP or Raspberry Pi.

damellis, 2016-09-09:


nebgnahz, 2016-09-15:
Can compile (there are quite some tweaks on the `CMakeLists.txt`, see (https://github.com/damellis/ESP/blob/chip/CMakeLists.txt).

When run, it's aborted.

```
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create
Aborted
```

With GDB, I get:

```
(gdb) r
Starting program: /home/chip/workspace/ESP/DEBUG/ESP
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0xb6a26e68 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
(gdb) bt
#0  0xb6a26e68 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
#1  0xb6a249e4 in OPENSSL_cpuid_setup () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
#2  0xb6fe2a7c in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0xbefff754, env=env@entry=0xbefff75c) at dl-init.c:72
#3  0xb6fe2b56 in call_init (env=<optimized out>, argv=<optimized out>, argc=<optimized out>, l=<optimized out>) at dl-init.c:30
#4  _dl_init (main_map=0xb6fff958, argc=1, argv=0xbefff754, env=0xbefff75c) at dl-init.c:120
#5  0xb6fd7a72 in _dl_start_user () from /lib/ld-linux-armhf.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Set `handle SIGILL nostop` will show:

```
Starting program: /home/chip/workspace/ESP/DEBUG/ESP
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create

Program received signal SIGABRT, Aborted.
__libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
47     	../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S: No such file or directory.
(gdb) bt
#0  __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
#1  0xb5a3e644 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#2  0xb5a3f34a in __GI_abort () at abort.c:89
#3  0xb5c18e8a in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb5c1789c in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#5  0xb5c178f2 in std::terminate() () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#6  0xb5c17b16 in __cxa_throw () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#7  0xb5c31d20 in std::__throw_length_error(char const*) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#8  0xb5c787c8 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned int&, unsigned int) ()
   from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#9  0xb5c78992 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#10 0x00493dca in of::priv::initutils() ()
#11 0x00401582 in ofInit() ()
#12 0x003edb32 in void ofSetupOpenGL<ofAppNoWindow>(std::shared_ptr<ofAppNoWindow>, int, int, ofWindowMode) ()
#13 0x003ed646 in void ofSetupOpenGL<ofAppNoWindow>(ofAppNoWindow*, int, int, ofWindowMode) ()
#14 0x003ec6f4 in main ()
```

nebgnahz, 2016-09-21:
The [rpi branch](https://github.com/damellis/ESP/tree/rpi) has the working code and configuration. To be merged into the master, work needs to be done on combining the `CMakeLists.txt` (shouldn't be too hard). Also `setup.sh` should download openFrameworks `armv6l` release.

Detailed diffs can be found [here](https://github.com/damellis/ESP/compare/rpi). Some of the changes are actually from [headless](https://github.com/damellis/ESP/tree/rpi) branch since they are needed to test the functionality on RPi. There are also changes that are available on the master but since this branch is rebased off the master, changes appeared again.

Note that only EGL is available on the embedded system, so we need the following patch that comments out OpenGL functions in `ofxDatGui`.
[ofxdatgui-rpi-patch.txt](https://github.com/damellis/ESP/files/484381/ofxdatgui-rpi-patch.txt)

damellis, 2016-10-13:
Merged in https://github.com/damellis/ESP/pull/392

closed 2016-10-13 by damellis