damellis/ESP #381: Remove video/ofQt* files from openFrameworks project.

damellis, 2016-09-23:
Mac OS Sierra (10.12) removes support for the QtKit that's needed by these files (resulting in the error `QuickTime/QuickTime.h: file not found`). We don't actually need those files for ESP, though, so we should remove them from the openFrameworks project, e.g. by munging the file in setup.sh.

damellis, 2016-09-23:
It looks like we can simply remove any line with "ofQt".

nebgnahz, 2016-10-07:
Re-opening and updated the title to track Mac OS Sierra build errors.

CMake build fails because in Mac OS Sierra. Error message:

```
ESP/third-party/openFrameworks/libs/openFrameworks/utils/ofConstants.h:66:14: fatal error: 'TargetConditionals.h' file not found
    #include <TargetConditionals.h>
             ^
1 error generated.
```

commit 43d66e9f (2016-09-23): Remove ofQt* files that depend on deprecated QtKit.

Fixes #381.
closed 2016-10-07 by nebgnahz