damellis/ESP #397: Question re: prediction output

narner, 2016-11-01:
Hi David and Ben, 

I'm working on using the Touché Arduino example + Processing in ESP mode. My question is regarding the order in which to run the Processing and Arduino examples in order to see the predicted class in the serial port, as you mention in [Step 9 in your Tutorial](https://create.arduino.cc/projecthub/mellis/gesture-recognition-using-accelerometer-and-esp-71faa1).

Currently, if I run Arduino with the serial monitor open, and then run ESP (Processing), I'm not able to select a serial port in ESP. Am I missing a step here? Thanks in advance!

damellis, 2016-11-01:
Great to see someone trying this out! Sorry you're having trouble...

You want to close the serial monitor in Arduino before running ESP. Opening the serial monitor is just to check that the Arduino is sending data, but it's not actually part of the ESP system's functioning. 

Let us know if you have any other questions!

narner, 2016-11-01:
@damellis My pleasure! This is a really fantastic project :)

Ok, closing the serial monitor makes sense; I had forgotten about that. So, based on Step 9; ESP does open a TCP connection that lets an Arduino read-in the class label values, correct? Is there an example showing that that I may have missed?

Also, would it make sense at all to send out the classification labels over OSC at all? I have some code that does that in OF I could provide (would need tweaking).

Thank you!

damellis, 2016-11-02:
The gesture recognition example sends the results over USB to the Arduino and over TCP to localhost port 5204. There's an example of reading the serial data (in Arduino) here: https://github.com/damellis/ESP/blob/master/Arduino/ADXL335_LEDs/ADXL335_LEDs.ino and the TCP data (in Processing) here: https://github.com/damellis/ESP/blob/master/Processing/BallDrop/BallDrop.pde

narner, 2016-11-02:
Looking at the Touché example, it looks like it should send the results out as well...I can see my ESP pipeline performing properly, but the results aren't affecting the BallDrop example. Looking at the BallDrop code, the results of `line` aren't being printed, which makes me suspect that `line` is `NULL`. Does the same thing happen to you?

damellis, 2016-11-03:
Hmm, not sure what's going on. The Touche example should also be sending data over TCP to Processing. You're seeing predictions being made within ESP?

narner, 2016-11-03:
I think it may have had something to do with the USB port I was using (?)
Here's a video of the running projects...I can now see that the BallDrop example is reading in the class label predictions, but the ball isn't on the screen:

https://www.youtube.com/watch?v=Hbep1gFXGlM

narner, 2016-11-13:
Hey guys, just wanted to check and see if you had a chance to take a look at the prediction over TCP error...thanks so much! Do let me know if there's any other information I can provide.

nebgnahz, 2016-11-15:
@narner I noticed in your video, the ball moves to the left at 14" when you start running the prediction. It makes sense because the predicted label is `1`. However, since it's all `1`, the `x` value keeps decreasing and becomes negative, making the ball invisible.

Try bound `x` or at least print `x` and see if that the cause. I think ESP and the TCP communication seems fine.

narner, 2016-11-20:
Ha, you're totally right @nebgnahz - I missed that completely, thank you.

closed 2016-11-20 by narner