damellis/ESP #398: Building the ESP project on Linux fedora Throw error for mpg123 library

MiladAlshomary, 2016-12-17:
Hello all,
I managed to build mostly everything for the project on my linux fedora system. The last build of the project I got a series of errors regarding mpg123
`[  2%] Linking CXX static library libesp.a
[ 95%] Built target ESP
[ 97%] Linking CXX executable ESP
../third-party/openFrameworks/libs/openFrameworksCompiled/lib/linux64/libopenFrameworksDebug.a(ofOpenALSoundPlayer.o): In function `ofOpenALSoundPlayer::initialize()':
ofOpenALSoundPlayer.cpp:(.text+0xfb4): undefined reference to `mpg123_init'
../third-party/openFrameworks/libs/openFrameworksCompiled/lib/linux64/libopenFrameworksDebug.a(ofOpenALSoundPlayer.o): In function `ofOpenALSoundPlayer::close()':
ofOpenALSoundPlayer.cpp:(.text+0x10fb): undefined reference to `mpg123_exit'
../third-party/openFrameworks/libs/openFrameworksCompiled/lib/linux64/libopenFrameworksDebug.a(ofOpenALSoundPlayer.o): In function `ofOpenALSoundPlayer::mpg123ReadFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<short, std::allocator<short> >&, std::vector<float, std::allocator<float> >&)':
ofOpenALSoundPlayer.cpp:(.text+0x194a): undefined reference to `mpg123_new'
ofOpenALSoundPlayer.cpp:(.text+0x196f): undefined reference to `mpg123_open'
ofOpenALSoundPlayer.cpp:(.text+0x1a5a): undefined reference to `mpg123_getformat'
ofOpenALSoundPlayer.cpp:(.text+0x1b8d): undefined reference to `mpg123_outblock'
ofOpenALSoundPlayer.cpp:(.text+0x1bf5): undefined reference to `mpg123_read'
ofOpenALSoundPlayer.cpp:(.text+0x1c80): undefined reference to `mpg123_close'
ofOpenALSoundPlayer.cpp:(.text+0x1c8c): undefined reference to `mpg123_delete'
.........`

I checked mpg123 and its installed on my system. After long search about this issue I found that I need to link the mpg123.a library statically. Any idea on how to do this?

Thanks!!

nebgnahz, 2016-12-17:
It seems you are using the CMake script to build the project.

If you know the exact location of your library `mpg123.a`, try modify the link libraries (see https://github.com/damellis/ESP/blob/master/CMakeLists.txt#L174) to include the path to the library. For example, 

```
target_link_libraries(${APP} PUBLIC
    ${PROJECT}
    <path to your mpg123.a>
)
```

Alternatively, the project configures libraries in an OS-dependent way. For *nix, we configure `SYS_LIBS` variable here: https://github.com/damellis/ESP/blob/master/CMakeLists.txt#L139. Snippet below:

```
set(SYS_LIBS "-L/usr/local/lib -lblas")
```

You may add the path to `SYS_LIBS`.

I don't have a Fedora to test, but if you have some luck, PR is welcome!

MiladAlshomary, 2016-12-18:
@nebgnahz I tried the first option you mentioned, I edited the linking section to be like this (I put my libmpg123.a file in the ESP project:

`target_link_libraries(${APP} PUBLIC ${PROJECT}
  ${PROJECT}
  ${ESP_PATH}/libmpg123.a
)
`

No am getting the following error:
` 90%] Building CXX object CMakeFiles/ESP.dir/Xcode/ESP/src/tuneable.cpp.o
[ 92%] Building CXX object CMakeFiles/ESP.dir/Xcode/ESP/src/main.cpp.o
[ 95%] Linking CXX static library libesp.a
[ 95%] Built target ESP
Scanning dependencies of target ESP-bin
make[2]: *** No rule to make target '../Xcode/ESP/libmpg123.a', needed by 'ESP'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 97%] Building CXX object CMakeFiles/ESP-bin.dir/Xcode/ESP/src/user.cpp.o
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/ESP-bin.dir/all' failed
make[1]: *** [CMakeFiles/ESP-bin.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

Any help please about what is going on?

closed 2016-12-18 by MiladAlshomary