damellis/ESP #399: Changing baud rate of the serial port

MiladAlshomary, 2016-12-22:
Hello,

Currently the ESP software reads serial ports on a baud rate  115200. How could I change this rate? Because my Lilypad XBee doesn't support this rate.

Thanks for help!

damellis, 2016-12-23:
You can try running ESP using the Processing mode: https://github.com/damellis/processing-esp-mode

This will allow you to view and edit the code for the examples, including the serial baud rate.

MiladAlshomary, 2016-12-23:
@damellis 

Thanks for your reply, I didn't find any installation instructions for the processing-esp-mode on Linux. Is there anyway to get it running on Linux? However I was able to run processing software on my Linux but couldn't link it with the processing-esp-mode.

Any help would be appreciated!

nebgnahz, 2017-08-11:
@MiladAlshomary If you are using the released binaries, there is no way to change the baud rate. The baud rate is unfortunately [hard-coded](https://github.com/damellis/ESP/blob/master/Xcode/ESP/src/examples/user_accelerometer_gestures.cpp#L6).

Other than using the Processing mode, you may try directly compile and run the project (see [instructions for developers](https://github.com/damellis/ESP/blob/master/README.md#installation-for-developers)).

MiladAlshomary, 2017-08-31:
Hey,
Yes I had to change the baud rate in the code and re build the project.

Thanks for your update

nebgnahz, 2017-08-31:
Closing for now.

closed 2017-08-31 by nebgnahz