damellis/ESP #400: Running the trained pipeline on the Arduino

giladnahor, 2017-06-18:
Hi, I'm trying to port the trained pipeline to run the classification on the Arduino itself (Arduino 101)
Is it possible? 
Training can be done offline but I want the project to be self sustained.
Thanks

nebgnahz, 2017-08-11:
Theoretically, the feasibility depends on the modules you use. For some pre-processing, feature extraction, and post-processing modules, they require more RAM than the embedded platform can offer. Also because of the slow CPU on embedded platforms, running prediction may be too slow.

Practically, porting GRT to embedded platforms may need to change quite some internal implementations. David (@damellis) has tried to [port GRT onto embedded platforms](https://github.com/damellis/grt/tree/arm). He can probably comment more.

damellis, 2017-08-13:
It would take a decent amount of effort to port the GRT to an Arduino 101. It also requires a full C++ implementation, enough to compile the STL; I'm not sure if the Arduino 101 has a full-featured enough compiler. There are also various changes to the GRT required to remove use of files, etc. There's a start to that in the port Ben linked to: https://github.com/damellis/grt/tree/arm