damellis/ESP #401: Errors when following "Installation for Developers" instructions

narner, 2017-08-05:
Hi all, 

I'm working on setting up ESP following the "Installation for Developers" instructions, rather than the Installation for Users instructions, which I've been using previously. I'm on macOS 10.12.5.

After running
```
git clone https://github.com/damellis/ESP.git
cd ESP
./setup.sh
```

I get this warning from the console:

```
ln: third-party/openFrameworks/addons/ofxGrt: No such file or directory
```

And, when opening the Xcode project and building, I get an error of

`ofxDatGui.h file not found`

Seems like an error in the setup script; though I'm not completely sure.

nebgnahz, 2017-08-05:
This looks like a submodule issue.

I think `clone` is not enough. It should have been `git clone --recursive ...` or performing `git submodule init; git submodule update` after the clone.

narner, 2017-08-05:
That makes sense - thanks for taking a look. 
I tried re-cloning using `git clone --recursive ..`, and then running `./setup.sh`. I'm still seeing what look like submodule errors, though: getting an `"ofxDatGui.h" not found` error. 

Looking at the submodule sources, it looks like there are a few missing files:

[image]

nebgnahz, 2017-08-06:
Mmm, after reading the setup script again, it looks like we've [taken care of the submodules](https://github.com/damellis/ESP/blob/master/setup.sh#L85).

The first error:

```
ln: third-party/openFrameworks/addons/ofxGrt: No such file or directory`---
```

suggests that the symlink fails. This is probably the same cause for all other missing files. Can you try creating [symlinks](https://github.com/damellis/ESP/blob/master/setup.sh#L88) yourself and see if it fixes your problem?

When we include add-ons to openFramework, they need to be in a specific location `${OF_ROOT}/addons`. Our `setup.sh` script first downloads openFramework, then clones custom addons, and then create symlinks (to avoid copying).

P.S. I just did a fresh clone / setup, and the script configures everything properly.

narner, 2017-08-06:
Hmm, Ok;  I *think* I fixed the symlinks, and rebuilding the project...however, now I'm getting some errors in the `ofApp.cpp` file: 

[image]

Sorry for all the questions...thanks for the help!

nebgnahz, 2017-08-07:
These are new functions introduced in https://github.com/damellis/grt/commit/e1cd0b57e401f4ec52ead679fb9926225585cf98 and they are not available in the original GRT source. Perhaps you are using an old version of GRT, e.g. you have previously installed GRT to the system (e.g. somewhere in `/usr/include`)?

narner, 2017-08-07:
> These are new functions introduced in damellis/grt@e1cd0b5 and they are not available in the original GRT source. Perhaps you are using an old version of GRT, e.g. you have previously installed GRT to the system (e.g. somewhere in /usr/include)?

Ahhh - yup, that's it. I have the original GRT installed for a few other projects...

I'm guessing it's not possible to have both simultaneously...I'll have to think about that. Any thoughts?

I can go ahead and close this issue, since the original problem has been resolved...

nebgnahz, 2017-08-07:
Try [changing the order of header search](http://docs.bluedotinnovation.com/display/DEVDOC10/iOS+-+Setting+the+Header+Search+Path).

- https://github.com/damellis/ESP/blob/master/Xcode/ESP/ESP.xcodeproj/project.pbxproj#L1121 for Debug mode
- https://github.com/damellis/ESP/blob/master/Xcode/ESP/ESP.xcodeproj/project.pbxproj#L1176 for Release mode

narner, 2017-08-08:
Definitely a good suggestion; I tried changing the order of the paths, but the project would then give me build errors indicating that it couldn't find *any* of the files for the GRT.

nebgnahz, 2017-08-10:
@narner Have you fixed it? If not, I will try tonight on my machine: install a released version of GRT in the system and change the header search order.

narner, 2017-08-10:
Hey Ben, no; hadn't been able to sort it yet - if you do end up testing,
I'll try out your solution straight away. Thanks again to you all for the
cool project!

nebgnahz, 2017-08-11:
I did a testing just now and was able to reproduce the error in [this comment](https://github.com/damellis/ESP/issues/401#issuecomment-320481533). After reorder the header search path (see commit https://github.com/damellis/ESP/commit/5e5268c4cd2a4e74438b8a5e56a397828f5879d8), I was able to compile and run.

---
My testing environment
Mac OS Sierra 10.12.6 (16G29)
Xcode Version 8.3.3 (8E3004b)
GRT: installed via `brew install grt`, which is installed in `/usr/local/Cellar/grt/0.0.1` and symlink-ed to `/usr/local/include`.

narner, 2017-08-11:
Awesome, that worked. I had to fiddle a bit with the symlinks, but I can build and run the project from Xcode. Thanks so much!

closed 2017-08-11 by narner